Ultimaker/Cura #19957: No Seam Corner Preference Cura 5.9

Btl64, 2024-11-26:
### Cura Version

5.9

### Operating System

Windows 10

### Printer

Elegoo Neptune

### Reproduction steps

Seam Corner Preference is disabled when User Specified is picked in Walls in Cur 5.9, but is available on previous versions, i.e., 5.7 & 5.8

### Actual results

Not able to choose Smart Hiding

### Expected results

I would expect it to be available like earlier versions.

### Add your .zip and screenshots here ⬇️

[image]
[image]

GregValiant, 2024-11-26:
Thanks for the report.
A lot of work was done to the Z-seam.  The new settings for "Scarf Seam" are in the Experimental section.
"Seam Corner Preference" is only available when "Sharpest Corner" is selected.  For a couple of the others "Z-Seam on Vertex" has taken it's place.  If you select "Random" then there is no option to pick a corner.

This looks like "expected behavior" as it is the way the settings are called out in fdmprinter.def.json.  I'll remove the bug label and close this.  I hope you understand.

closed 2024-11-26 by GregValiant

Btl64, 2024-11-27:
I have a couple of models that I was able to create amazing seams when
sliced with the earlier versions in the way I described.  I haven't been
able to replicate the look with 5.9.  I'll just keep using the older
versions.

GregValiant, 2024-11-27:
The setting "Zseam on Vertex" was supposed to take it's place.  That one seems odd to me.
I've played with "Scarf Seam" and after some trial and error it appears to be working well.  They don't stick out which makes mating parts (like threads in particular) fit together better.