Ultimaker/Cura #19959: Head is moving to MAXX and MAXY before printing

fixundfertig123, 2024-11-27:
### Cura Version

5.8.1

### Operating System

Win 11

### Printer

Bibo2

### Reproduction steps

Unfortunately a couple of month ago (I think after some cura3d update) all my newly created print jobs contain a fast G0 movement to MAXX and MAXY prior to printing. I have not enabled prime tower. The reason for the movement is caused by a line inserted at the beginning between the layer count output:
....
;LAYER_COUNT:26
**G0 F7200 X210.575 Y162.575**
;LAYER:0
.... 

This issue has been confirmed and discussed [at the forum](https://community.ultimaker.com/topic/46354-why-is-print-head-moving-to-maxx-and-maxy-before-printing/). 

### Actual results

```
;FLAVOR:Marlin
;TIME:946
;Filament used: 0.868348m, 0m
;Layer height: 0.2
;MINX:45.064
;MINY:76.301
;MINZ:0.25
;MAXX:210.575
;MAXY:162.575
;MAXZ:20.001
;TARGET_MACHINE.NAME:BIBO2 
;Generated with Cura_SteamEngine 5.8.1
T0
M140 S80
M105
M190 S80
M82 ;absolute extrusion mode
M109 T0 S265.0
G21                     ;metric values
G90                     ;absolute positioning
M107                    ;start with the fan off
G28 X0 Y0               ;move X/Y to min endstops
G28 Z0                  ;move Z to min endstops
G1 Z2.0 F400            ;move the platform down 2mm
T0
G92 E0
M75                     ; Start Print Job Timer
M117 BIBO Printing...
G1 F1500 E-6.5
;LAYER_COUNT:26
G0 F7200 X210.575 Y162.575
;LAYER:0
...
```

### Expected results

```
;FLAVOR:Marlin
;TIME:946
;Filament used: 0.868348m, 0m
;Layer height: 0.2
;MINX:45.064
;MINY:76.301
;MINZ:0.25
;MAXX:210.575
;MAXY:162.575
;MAXZ:20.001
;TARGET_MACHINE.NAME:BIBO2 
;Generated with Cura_SteamEngine 5.8.1
T0
M140 S80
M105
M190 S80
M82 ;absolute extrusion mode
M109 T0 S265.0
G21                     ;metric values
G90                     ;absolute positioning
M107                    ;start with the fan off
G28 X0 Y0               ;move X/Y to min endstops
G28 Z0                  ;move Z to min endstops
G1 Z2.0 F400            ;move the platform down 2mm
T0
G92 E0
M75                     ; Start Print Job Timer
M117 BIBO Printing...
G1 F1500 E-6.5
;LAYER_COUNT:26
;LAYER:0
...
```

### Add your .zip and screenshots here ⬇️

-
[cura.log](https://github.com/user-attachments/files/17932958/cura.log)

GregValiant, 2024-11-27:
Thanks for the report.
I played with this some more.
(BTW The behavior is the same with an Ultimaker S5 active.)
The OP's problem is the travel-to-the-prime-tower line that Cura inserts at the end of the StartUp Gcode section.
;LAYER_COUNT:348
**G0 F7200 X97.575 Y63.575**
;LAYER:0

In your "BIBO" extruder definition files, in the "overrides", are these lines:
"machine_extruder_end_pos_x": { "value": "prime_tower_position_x" },
"machine_extruder_end_pos_y": { "value": "prime_tower_position_y" },
"machine_extruder_start_pos_x": { "value": "prime_tower_position_x" },
"machine_extruder_start_pos_y": { "value": "prime_tower_position_y" },
A workaround is to change them to:
"machine_extruder_end_pos_x": { "value": "prime_tower_position_x if prime_tower_enable else 0" },
"machine_extruder_end_pos_y": { "value": "prime_tower_position_y if prime_tower_enable else 0" },
"machine_extruder_start_pos_x": { "value": "prime_tower_position_x if prime_tower_enable else 0" },
"machine_extruder_start_pos_y": { "value": "prime_tower_position_y if prime_tower_enable else 0" },
That will eliminate that travel line unless the Prime Tower is enabled.
You could also delete those lines, or change the endings to: { "value": 0 } which effectively disables them.

I see some other problems in Cura in regards to those settings.

- The printhead always moves to "Extruder Start Prime Position" whether or not the prime tower is enabled and regardless of the extruders_enabled_count or the number of extruders used in the print.
- The printhead never moves to "Extruder End Prime Position".
- Neither the "machine_extruder_start_pos_x" nor the "machine_extruder_start_pos_y" can be enabled in the Cura settings.  So they cannot be changed by the user, only from a definition file.

Someone from the Cura team will take a look.

fixundfertig123, 2024-11-28:
HI @GregValiant. Thanks again for taking a look at my issue that quick! I really appreciate it. 

I validated your suggestions and this workaround works great for me and saves me quite some time, since otherwise I have to manually remove the line. And I forget to do it frequently ;-)

Secondly I tried to reproduce my problem and figure out when it started. I freshly installed different versions of Cura. In 5.7.0 and the extra line of code occurred. However, if I am not mistaken in 5.6.2 this line was not added. I hope that I did not pickup any legacy setting somewhere. However this observation looks interesting and I wanted to let you know for tracing down the problem.

GregValiant, 2024-11-28:
I went back to 4.13.1 and the "start" position seemed to work correctly, but the "end" position never showed up in the gcode.
The Prime Tower was reworked a couple of versions ago when the "Interleaved" option was added.  It is not unusual to get "unexpected consequences" in the code when making significant changes to existing code.
Glad the workaround helped as the Cura team has been busy with other things so it will likely be a while before this gets addressed.

fixundfertig123, 2024-11-29:
I'm familiar with kind of side effects :-). And it is absolutely fine with me to not being prioritized since there is a workaround. Thanks again for all your work.

fixundfertig123, 2024-12-01:
Hi. Thank you for your quick reply. I am by far not an expert, and I do understand your intention. However, in cases where there is no prime tower (imho in most single printhead & one color printing), there is simply no cleaning taking place. On the contrary, things potentially are getting worse because you are dragging any potential nozzle oozing across the clean but hot&sticky build plate. And how do you handle the case you want to print multiple jobs in different build plate areas without removing the prior objects? And why was it introduced in release 5.7.0 without special notice? Therefore, I personally follow @GregValiant 3-point argumentation:

- The printhead always moves to "Extruder Start Prime Position" whether or not the prime tower is enabled and regardless of the extruders_enabled_count or the number of extruders used in the print. 
- The printhead never moves to "Extruder End Prime Position".
- Neither the "machine_extruder_start_pos_x" nor the "machine_extruder_start_pos_y" can be enabled in the Cura settings. So they cannot be changed by the user, only from a definition file.

Since there is workaround I am happy now. Nevertheless - not at high priority - I would suggest to look at and implement @GregValiant clean suggestions.

fixundfertig123, 2024-12-03:
Dear @HellAholic. Thank you for your time to explain your point of view. I really understand your point and intention. 

Regarding you pictures: the effect really depends upon you homing position. In the second picture, if your homing would be in the lower left corner, and if you would have nozzle oozing during heat up, your move would have dragged the nozzle oozing over your future print as well.

And maybe it is just a wording and general thing on how you view on this issue. You yourself say: 
_"...
You can switch the prime tower position X and prime tower position Y from within Cura which will update the move.
There is a setting that was added in 5.8.0 which is called Layer Start X and Layer Start Y which inherits from the calculated location for the move (prime tower position in this case), 
..."_
Please do not get me wrong. For me as an end-user (and I am an software developer myself) is is quite confusing and not very clean to have a independent setting that inherit from disabled feature settings and therefore enabling this disabled feature actuator moves. If I could make a suggestion I would introduce two independent variable settings which can independently be enabled / disabled:

1. Start position after homing:
    - Enabled / Disabled
    - Start X / Start Y Position
    - ...
2. Prime Tower:
    - Enabled / Disabled
    - Start X / Start Y Position
    - ...

fixundfertig123, 2024-12-27:
Dear @GregValiant. Sorry, I was a away project-wise. Thank you for picking it up and all your effort. I am curious to test it once published.

GregValiant, 2024-12-27:
You can test it now.  It could use another test driver.
Unzip and put "PurgeLinesAndUnload" into the "Configuration Folder" and "scripts" sub-folder.  It will be available with the other post processors.
[PurgeLinesAndUnload.zip](https://github.com/user-attachments/files/18260305/PurgeLinesAndUnload.zip)

There are 4 options (Add Purge Lines, Move to Layer Start, Adjust Starting E, and Unload Filament).
Rather than being an option - "Move to Prime Tower" is automatic and always runs.  It checks to see if that prime tower move is in the gcode, and if it is, it adds G0 moves to the periphery of the build plate so the nozzle doesn't cross the print area.  It eliminates that string that can get pulled across.

wawanbreton, 2025-02-11:
Hi, I'm doing some investigation about this issue. The value `"machine_extruder_end_pos_x": { "value": "prime_tower_position_x" },` really doesn't make sense, as pointed out by @GregValiant . The description says that this is "The x/y-coordinate of the starting position when turning the extruder on", so there is no way this can be dynamic. It should be a fixed value, defined by the manufacturer, which indicates the position of the nozzle at the and of the printer initialization sequence, just before the G-Code start being processed. Then the engine will try to optimize its moves according to this original position, and avoiding trailing ooze wires.

I think the confusion comes from the definition of the ultimaker_original, which is linked to the prime tower position, and then propagated to newer definition files. But for all more recent UltiMaker printers, we do have the actual fixed start position. 

We could decide to just remove those definitions, and the default would then be 0/0, but this may help in some cases and make others worse. The only way to properly fix this is to set the actual fixed value, by printer. So if you are able to find the proper start position for your model, please make a PR to set the actual start position, or at least send it to us directly so that we can integrate it.