Ultimaker/Cura #19962: Not poping up the file when i insert it into the cura

Benobrody, 2024-11-27:
### Cura Version

5.4.0

### Operating System

Mac os

### Printer

Elegoo Neptune 4 Plus

### Reproduction steps

1. Put in the file to slice it nothing showed up 2. deleted software and re-downloaded it didn't work .

### Actual results

it still wont do it

### Expected results

give me a way to fix it.

### Add your .zip and screenshots here ⬇️

[skull bank - 4179550.zip](https://github.com/user-attachments/files/17936513/skull.bank.-.4179550.zip)

GregValiant, 2024-11-27:
Thanks for the report.
That model has a lot of errors.  The slot in the top is partially closed, and I don't see a way to get the money out without smashing the print.
I'll mark this as a duplicate report and close it as 5.4 to 5.6 had well known slicing problems.  I hope you understand.

This is in 5.9.0 and at 50mm/sec it's a 54 hour print.  The top of the open area inside is flat and will need supports which you won't be able to get out.
[image]

closed 2024-11-27 by GregValiant

Benobrody, 2024-12-21:
It didnt work

GregValiant, 2024-12-21:
Which part didn't work?
Did you upgrade to 5.9.0?  That will help, but that's a really bad model.  It really needs to be reworked in CAD.