Ultimaker/Cura #19963: Cura not slicing at 0.20mm layer height, was working fine before (no settings were changed)

amgrey70, 2024-11-27:
### Cura Version

5.9.0

### Operating System

Windows 11

### Printer

Anycubic Kobra

### Describe your model

I didn't change any settings when slicing this, it slices fine when layer height is 0.15mm, but 0.16, 0.17 etc. up to .20 result in an unable to slice error.

### Add your .zip here ⬇️

[AK_Ornaments.zip](https://github.com/user-attachments/files/17937774/AK_Ornaments.zip)

GregValiant, 2024-11-27:
Thanks for the report.
I didn't have any trouble slicing it in 5.9.0 on my Win10 laptop.  I didn't find any errors in the models.
Use the "Help | Show Configuration Folder" command to locate the Cura.log file.
Recreate the "won't slice" problem and then go to the config folder and zip the log file and post that zip folder here.  Maybe it will give a clue as to what happened on your system.

amgrey70, 2024-11-27:
That's okay, I think it may have been the large ornament file's problem. I'm just using a different model now, but I thought you guys might want to know about the error. Thank you so much for the super fast response, and have a happy thanksgiving!