Ultimaker/Cura #19965: 100% Infill with Concentric

Quantun97, 2024-11-27:
### Cura Version

5.9.0

### Operating System

Windows 11

### Printer

Ender 3

### Reproduction steps

I got a benchy stl file and did 100% Concentric. After slicing under "Preview" it shows the benchy with lines instead of Concentric lines.

Now the settings still show concentric but the preview says otherwise after slicing because the benchy shows lines as the infill.

### Actual results

It did 100% infill with lines instead of Concentric

### Expected results

100% infill with Concentric

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2024-11-28:
Thanks for the report.
I can't duplicate this.  Please post a project file.
This is in 5.9.0.
[image]

Quantun97, 2024-11-28:
How will I go about posting a project file here? STL nor OBJ file types don't work.

Asterchades, 2024-11-28:
This isn't a bug - just a peculiar default.

By absolute default (so no intervention at all by definition or user), when you set the *Infill Density* to 100% the *Bottom Layers* count is set to 999,999. This ultimately yields the same result in that the final print is solid, but it does so by making entirely out of skin rather than infill. This is why your preview is showing the yellow *Top / Bottom* colouration rather than the expected orange *infill* colour of Greg's, as Greg has overridden this default behaviour.

This does mean that it is going to change the print properties. Infill is generally printed much faster than Skin (twice as fast by default), can be done with a different pattern (although this is moot at 100% density), can be printed at a greater effective layer height using the *Infill Layer Thickness* setting, and doesn't necessarily have to be printed using the same extruder.

Note that this also sets the *Top Layers* to 0, which will interfere with the *Top Surface Skin Layers* feature.

Quantun97, 2024-11-28:
Thank you for the thorough explanation. I appreciate the help

Asterchades, 2024-11-28:
I should have also mentioned that you can easily override this on your own. You can either manually re-set the _Top Layers_ and _Bottom Layers_ value, use a very-slightly-under 100% _Infill Density_ - like 99.999% - as the behaviour will only trigger at exactly 100%, or set the _Infill Line Distance_ to match your _Infill Line Width_ as this gives 100% density without saying exactly that.

Quantun97, 2024-11-28:
Good to know. I will be giving it a try.

Quantun97, 2024-11-28:
What an interesting default. I am now however able to print at 100% infill by changing those settings.

closed 2024-11-28 by Quantun97