Ultimaker/Cura #19966: Fix bad GCODE generation

Bfitzy, 2024-11-27:
### Is your feature request related to a problem?

It really bothers me when I go to print something and it waits for the bed to heat up before heating up the extruder, when the extruder can be heated up while the bed is heating. When i'm printing back to back its really annoying because the extruder has to cools down while the bed is warming up and then has to take a few extra minutes to heat up when it could just be held at a higher temperature, and I can't seem to find a way to change this behavior.

### Describe the solution you'd like

have an option to preheat the bed first or preheat both at the same time when the print job starts ( select the order of the M104 and M140 commands)

### Describe alternatives you've considered

trying to write custom GCODE, but the M104 and M140 commands are auto generated by cura

### Affected users and/or printers

all, probably

GregValiant, 2024-11-28:
This is printer dependent.  I think the UM printers handle this in the firmware.  For other printers it is handled by the StartUp Gcode.  It's just a question of the command order.  The power supplies of less expensive printers may not be able to handle the initial power load of both heaters starting at once.  That's the reason it isn't an option in Cura.
Cura 5.9.0 introduced some logic opportunities into the StartUp and Ending gcodes.  The examples here are suitable for 5.9.0.

If you don't have any temperature lines in your StartUp, then Cura adds the "prepend temperatures" right after the "Generated with..." line in a gcode file.  Those lines are affected by the settings "Wait for build plate heatup" and "Wait for nozzle heatup".  Those are available if you load the "Printer Settings" plugin from the MarketPlace.

If Cura sees ["replacement keywords"](https://files.fieldofview.com/cura/Replacement_Patterns.html) for the temperatures in the StartUp then those prepend temperatures that you don't like won't be added.
If your StartUp Gcode is configured something like this then the bed and heater will start together.
M104 S{material_print_temperature_layer_0} ; start the hot end
M140 S{material_bed_temperature_layer_0} ; start the bed
M109 S{material_print_temperature_layer_0} ; wait for the hot end
M190 S{material_bed_temperature_layer_0} ; wait for the bed

But if you have an ABL you might want it to be different so the nozzle doesn't ooze during the leveling travels.
M104 S170 ; start the hot end
M190 S{material_bed_temperature_layer_0} ; wait for the bed
G28 ; auto-home
G29 ; auto-level command
G0 F{speed_travel*60} X0 Y0 Z15 ; move to the corner to ooze
M109 S{material_print_temperature_layer_0} ; wait for the hot end

Personal preference comes into play as well.
This is the startup on my Ender 3 Pro with no ABL.  The E-steps/mm are adjusted depending on the material.  The bed waits if it is set to 65 or higher (when printing PLA I don't bother to wait for the bed).
;--------------------------StartUp Gcode
M113 S0                    ; No Echo's
{if 'TPU' in material_type}
M92 E119                   ; Steps/mm for TPU
{elif 'PETG' in material_type}
M92 E102                   ; Steps/mm for PETG
{else}
M92 E97}                   ; Steps/mm for PLA
{endif}
{if machine_center_is_zero}
M206 X-116 Y-121 Z0     ; Home offsets 'Origin at center'
{else}
M206 X-1 Y-6 Z0         ; Home offsets Origin LF corner
{endif}
G21                        ; Metric Units
G90                        ; Absolute Movement
M82                        ; Absolute Extrusion
M220 S100                  ; Reset FeedRate
M221 S100                  ; Reset FlowRate
M140 S{material_bed_temperature_layer_0}                   ; Start heating the bed
{if material_bed_temperature_layer_0 > 65}
M104 S200                  ; Standby while bed heats
M190 S{material_bed_temperature_layer_0}                   ; Bed wait if over 65
{endif}
M109 S{material_print_temperature_layer_0}                 ; Hot End Temperature
G92 E0                     ; Reset E
G28                          ; Home all axes
G0 F{speed_z_hop*60} Z2                 ; Move up
;--------------------------End of StartUp

What you are asking for has been possible in Cura since at least version 4.4.  The Creality, Elegoo, and other printers come with the stock StartUp Gcode configured to heat the hot end and bed together.
I'll go ahead and close this.  Let me know if you think that is premature.

closed 2024-11-28 by GregValiant

Bfitzy, 2024-11-28:
That was an awesome reply. Thank you so much for explaining this so well. I hope you have a wonderful Thanksgiving!Brandon Fitzgerald