Ultimaker/Cura #19969: Cura 5.9 fails to slice

mralexc, 2024-11-28:
### Cura Version

5.9

### Operating System

Linux

### Printer

FLSun V400

### Describe your model

Tool battery holder

### Add your .zip here ⬇️

I tried multiple .stl files and after the update to 5.9 I am no longer able to slice.
[Milwaukee M18 battery holder - 3983547.zip](https://github.com/user-attachments/files/17950096/Milwaukee.M18.battery.holder.-.3983547.zip)

mralexc, 2024-11-28:
[cura.log](https://github.com/user-attachments/files/17950149/cura.log)

GregValiant, 2024-11-29:
Thanks for the report.
I didn't have any trouble slicing them on my WIn10 system.  The models don't appear to have any errors in them.
There are a lot of errors and warnings in the log file.  This is a crash in the Cura Engine but I don't know the crash codes.
_Backend exited abnormally with return code -11!_
Someone from the Cura team will take a look.

mralexc, 2024-11-29:
Yeah, I saw some of that in the log that looked odd, but not fatal. Also to confirm, I ran the slice successfully in 5.8.1 as a test. Thanks for taking a look. This is Ubuntu 22.04

mralexc, 2024-11-29:
Is this a new setting? Something that needs to be added into my variant config?

tbayart, 2025-03-19:
Hi, i have the same issue since i updated from Cura 5.6 to 5.9.1
The model i'm trying to slice were previously perfectly sliced under 5.6 without any issue

[cura.log](https://github.com/user-attachments/files/19343513/cura.log)
[Kunai support.zip](https://github.com/user-attachments/files/19343485/Kunai.support.zip)

Let me know if you want more information or if you prefer i open a separate issue

GregValiant, 2025-03-19:
A separate report would be best.  Sometimes it's the model, or a Cura bug, or the user settings.  It's best to keep them separate here