Ultimaker/Cura #19974: Print volume representation is not matching the real printer's printing volume

FX2LTD, 2024-11-30:
### Cura Version

5.9.0

### Operating System

Windows 11

### Printer

Elegoo Neptune 4 MAX

### Reproduction steps

I have selected in the Printer Properties the ELEGOO NEPTUNE 4 MAX, and after a few seconds that the printer was imported, the blu volume height lines became very short. I cannot print anything above a certain height because the software refuses to slice the object.

### Actual results

I cannot slice and upload the file to the printer

### Expected results

The printing volume should be big enough to host the whole object

### Add your .zip and screenshots here ⬇️

[Uploading EN4MAX_1122 Batman shin right.zip…]()

GregValiant, 2024-11-30:
Thanks for the report.
The project file didn't upload correctly so I can't check your machine settings.  Please try again.

In "One at a Time" mode the height of the print volume is limited by the Gantry Height.  Go to the "Machine Settings" and in the right pane will be the Gantry Height settings.  You definition file says it should be 462.0.
Also in the Machine Settings is X(width), Y(depth), and Z(height) settings.  Yours should be 430 x 430 x 482.

FX2LTD, 2024-11-30:
[image]
This is my Elegoo Neptune 4 Max settings page, the way it is now, without me having touched something....
It seems exactly how you mentioned the settings should be. What I do not understand is why the parallelepiped appears for a second, then shrinks to a shorter size.... I will try to uninstall the software completely and reinstall it, but I have the same issue on another computer... I will see if I can make a video of that short time in which the volume shrinks... The project size I am not sure it has anything to do, as I have this situation whether I load any file into the software or not...

FX2LTD, 2024-11-30:
Excuse my ignorance... Where do I find "Print Sequence" set to "One at a Time", please?

FX2LTD, 2024-11-30:
I found "Print Sequence" in the settings... Nothing seeems to have changed. In what format should I save the project? Universal Cura or .3mf?

FX2LTD, 2024-11-30:
[Uploading EN4MAX_1122 Batman shin right.zip…]()

GregValiant, 2024-11-30:
Once again the link just bunces me back to this page.
Let's go the other way.
Unzip the file and open it in UM Cura.  The printer is an Elegoo Neptune 4 Max and will overwrite your printer.  Hopefully it will straighten things out.
[EN4Max_35 Cube.zip](https://github.com/user-attachments/files/17965359/EN4Max_35.Cube.zip)

FX2LTD, 2024-11-30:
> Once again the link just bunces me back to this page. Let's go the other way. Unzip the file and open it in UM Cura. The printer is an Elegoo Neptune 4 Max and will overwrite your printer. Hopefully [...]

Unfortunately it did not work: Two columns opened into the working space, the gantry height is still low...
I could see the gantry borders going up and again down, during the project download. As this is happening on three different computers, one with Windows 10, the other two with Windows 11, I will try to completely uninstall on one computer and reinstall.
On Orca slicer and the Creality slicer it does not happen. On Elegoo Cura 5.6.0 it happens...
I do not know what to do in order to upload files correctly... I have zipped the video file and clicked on the "Paste, Drop, or click to add files" icon below the message box...
[20241130_190209_edit1.zip](https://github.com/user-attachments/files/17966507/20241130_190209_edit1.zip)

FX2LTD, 2024-11-30:
Hi,
I think I have managed to find a partial solution, thank you to your file. I have realised that, if your file was still not able to change anything, it could be a narrower issue. So I have uninstalled the software on another computer, cleaned the registry, wiping out every possible trace of information about the software, and I have reinstalled it, after rebooting the system. As soon as the software loaded, it caught the printer immediately and showed me some messages about some additional plug-ins. Then I thought that the issue might not have been host-related, if it was happening on different machines, but on something that was linked to my online account. I do not cura well enough to know exactly what is being stored locally onto the machine, and what is actually loaded in realtime from the internet. So, I have uninstalled or disabled all the plug-ins, and when I have reloaded the Cura software, the problem was gone.
I have re-enabled all the plug-ins, and the problems is not re-occurring. I have tried to disable/enable the plugins in groups of 2-3 at a time, but I could not narrow down the issue to any specific plug-in.
For now I have managed to fix the issue, so I could say it belongs to some third-party software, but I cannot tell which one.
Thank you for all your efforts.
Regards

closed 2024-11-30 by FX2LTD