Ultimaker/Cura #19980: Cura 5.8 vs 5.9: 5.9 slicing produces extraneous overhangs

cytan299, 2024-12-01:
### Cura Version

5.9.0

### Operating System

MacOS Sequoia 15.1.1

### Printer

Ender3 v3 SE

### Reproduction steps

I have the same 3mf that I use in both 5.8.0 and 5.9.0. I sliced in 5.8 and 5.9 and sent those files to my printer. The 5.9 gcode produces extra overhangs while 5.8 produces nice and smooth edges. See photo in "Actual Results" below. Project file is enclosed plus the gcodes that were generated by both 5.8.0 and 5.9.0

### Actual results

See photo with overhangs marked. The two printed objects are compared using the same printer. The only difference is that the one with the overhangs was created with 3.9.0 while the one without is with 3.8.0.
[image]

### Expected results

The photo of the expected results are shown in "Actual results".

### Add your .zip and screenshots here ⬇️

Zip file (bugs.zip) contains:
* _test.3mf_:  project file used for in both 3.9.0 and 3.8.0 for slicing.
* _test_5.8.gcode_: gcode created by CURA 5.8.0 with the _test.3mf_ project file.
* _test_5.9.gcode_: gode created by CURA 5.9.0 with the _test.3mf_ project file.

[bugs.zip](https://github.com/user-attachments/files/17967441/bugs.zip)

GregValiant, 2024-12-03:
Thanks for the report.
I have some code analyzing tools I use.  I read both gcode files into AutoCad and they are alike as two peas in a pod.

I know this is hard to see but I have overlayed the outer walls of the 5.8 file over the outer walls of the 5.9 file in AutoCad.  Everything is spot on.  The toolpaths are identical.
[image]

BUT the 5.9 file used almost 20% more filament (.739m to .614m) and my slice of your project shows it using .648m.  Maybe you had made a flow adjustment?

Those aren't "overhangs" in your images.  The project file shows you have the Z-seam in the front so what you are likely looking at is sloppy starts and stops to the extrusions.

Looking at the model, it has a couple of small errors but I'm pretty sure it isn't a problem.  I think if I was going to print this I'd drop down to at least a 0.3 nozzle and maybe all the way to 0.2 (which my Ender is not fond of).  Those little "pillars" on the sides are a problem with the .4 nozzle.  The toolpath around those pillars is only 0.24mm radius.  The layers are very short duration and cooling is a real problem with this model.  What you are seeing as bad Gcode may just be a layer cooling problem.

I printed your 5.9.0 gcode file (with some funky leftover filament).  It doesn't show the problems that your print does.  I don't know what to make of that.  I consider this a pretty poor print but as I say, I'd do it with a smaller nozzle.
Someone from the Cura team will take a look.

[image]

[image]

cytan299, 2024-12-03:
The weird thing is that I thought that it was a one off,  but I reproduced the problem multiple times with the 5.9.0 gcode, i.e. more than 3. Did you try to print the 5.8.0 gcode to see if there's any difference with the results?

cytan