Ultimaker/Cura #19984: Gaps during print

luizbgomide, 2024-12-01:
### Cura Version

5.9.0

### Operating System

Windows 11

### Printer

TwoTrees Bluer

### Reproduction steps

Slice the following model: [CoC - Right-Expansion.zip](https://github.com/user-attachments/files/17969898/CoC.-.Right-Expansion.zip)

### Actual results

I've sliced the and notice gaps on the front left corner and in the second wall from the left:
[image]
[image]

Those are the only two places with gaps, I slided a [minimum test.zip](https://github.com/user-attachments/files/17969925/minimum.test.zip) (the model sunken 20 mm) and test many different variables, from Flow to Junction Deviation and Linear Advance to Layer Height (.2 to .32) with no visible changes (I had to increase Flow by 10% to start to close the gap, but since I only had gaps on two spots I don't think Flow is culprit here).

### Expected results

No gaps everywhere. I'm not sure if the issue is mechanical or from the slice, but since I've only seem this in a couple of parts, in some especific places near curves and only recently (with the latest version of Cura) I suspect there might be something off with the slicing.

### Add your .zip and screenshots here ⬇️

The sliced section looks just fine on Cura 
[image]

GregValiant, 2024-12-02:
Thanks for the report.
We'll need a project file ("File | Save Project") to investigate.  Zip the project file and post it here.
I haven't seen this problem in my own prints with 5.9.0.
The walls are 1.6mm thick and so a 0.40 line width should fill them just right.  With my settings; the highest flow rate I see is 4.25mm³/sec which is pretty low.

luizbgomide, 2024-12-02:
The file `minimum test.zip` on Actual Results section is a 3mf project file.