Ultimaker/Cura #19986: Slicing is failing on voronai lamp print

JaxPratt, 2024-12-03:
### Cura Version

5.9.0

### Operating System

Windows 11

### Printer

Creality CR-10

### Name abnormal settings

0.25mm extruder height quality with a .4mm extruder

### Describe model location

It slices if I rotate it 180 degrees.

### Describe your model

It only sliced once on it's original print orientation. 

### Add your .zip here ⬇️

[CCR10_tmpwi_lib9cvoronoi_pearl_lamp_no_1_ms_105mm_A-shell.zip](https://github.com/user-attachments/files/17985433/CCR10_tmpwi_lib9cvoronoi_pearl_lamp_no_1_ms_105mm_A-shell.zip)

GregValiant, 2024-12-03:
Thanks for the report.
I sliced it three times and didn't have a problem.
Set Cura up and recreate the problem.  Then, use the "Help | Export Package for Technical Support" command and post the zipped file here.  It will contain your log file.  Maybe that will give a clue.

GregValiant, 2025-03-11:
There was never a response to this so I'll close it.  Let me know if you feel that was premature.

closed 2025-03-11 by GregValiant