Ultimaker/Cura #19988: Cura 5.9 crashes when opening file

lozdown, 2024-12-03:
### Cura Version

5.9

### Operating System

Ubuntu  24.04

### Printer

Anycubic I3 Mega

### Reproduction steps

1 Started linux appimage normally
2 Open stl file
3 Very slow to open but does eventually
4 Slice and export to gcode file (which prints perfectly)
5 Clear build plate
6 Open another stl file
7 File selection dialog starts to open
8 Cura disappears and has to be restarted

### Actual results

Cura crashes

### Expected results

Should continue to work normally

### Add your .zip and screenshots here ⬇️

[cura-bug.zip](https://github.com/user-attachments/files/17990911/cura-bug.zip)

xsaMNJfkltrwq5r3, 2024-12-06:
Using Ver. 5.9.0 but also occurred on previous versions.  
Opening a file takes a long time (approx 5mins sometimes).  Does eventually load onto build plate.
Searching for fix online, it was suggested to disable USB Printing in Marketplace.  This completely fixed this issue for me.  Re-enabling caused the issue to reappear, then disabling and restarting cleared the issue.

DySprozin, 2025-01-12:
@xsaMNJfkltrwq5r3 , oh, thank u very match! 

> Opening a file takes a long time (approx 5mins sometimes)
yes, i had this problem too

>  disable USB Printing in Marketplace

its working!