Ultimaker/Cura #19990: Please Add Top/bottom layer 'thickness'/height

fast2611, 2024-12-03:
### Is your feature request related to a problem?

The "Infill Layer thickness" , "Support Infill Layer thickness" and "initial layer height" settings have been godsends for both increasing part strength and print speed, and are frankly the only reason why I still use cura for all my projects.
However, there is no such setting for what would most definitely be the biggest time saver of them all, the top and bottom layers.

### Describe the solution you'd like

Id like to either have a "Top/botton Layer thickness" setting, or a "Top/botton infil Layer thickness" added wich respetivly would increase the layer height of the fiil pattern, or the entire line type, rounding to the current layer height if not an exact multible of it (i.e. 0.28 rounds to two layers if you use .12 or 1.6 layer height both) the existing settings already work like that too.
The reason this setting should exist is because it allows heigh detail on the outside of the print and high print speeds, while maintaining thick, sturdy strands of plastic on the inside

### Describe alternatives you've considered

You are making me fill the alternatives section so... idk? maybee there is a different slicer that has this feature? or I could pay someone to make it a marketplace plug-in? Frankly id be happy to do either for how much electricity and time this would save

### Affected users and/or printers

All of them, but ones with high flow rates would benefit more

LilBub, 2024-12-03:
Interesting suggestion.  I hope they consider it

fast2611, 2024-12-03:
I believe the "infil" top bottom layers use is called skin? so "Top/botton infil Layer thickness" should probably be called "Top/botton skin Layer thickness". Though I frankly do not care what its called as long as I get some setting that lets me do this