Ultimaker/Cura #19992: Faster printing for multiple objects: semi-all-at-once (switch object every N layers)

x3e, 2024-12-03:
### Is your feature request related to a problem?

With current slicers, I don't like the way multiple objects are printed. When printing all-at-once, a lot of time is spent on travelling moves between the objects. With one-at-a-time, the object size is severely limited because the head may bump into other objects when they are already fully printed.

### Describe the solution you'd like

It would be great if the printer could print a few layers (for example 5) of each object, then go over to the next object, Lower the extruder/bed five layers again, then print five layers, then move over to the next object etc. It would save a lot of moves and thus print time. This way, you could print multiple objects very fast, while only being constrained by the nozzle size (and in my case, the fan on the left side of the nozzle. This is could be solved by starting to printing objects from right to left, or, keeping a bit of space between the objects).

so instead of
object 1 - layer 1
object 2 - layer 1
object 1 - layer 2
object 2 - layer 2
object 1 - layer 3
etc.

I woud like this:
object 1 - layer 1
object 1 - layer 2
object 1 - layer 3
object 1 - layer 4
object 2 - layer 1
object 2 - layer 2
object 2 - layer 3
object 2 - layer 4
object 1 - layer 5
etc.

### Describe alternatives you've considered

As mentioned, printing one-at-a-time or all-at-once (the current print modes) work, but are slower.

I might be able to implement the behaviour with a post-script or plugin that interleaves layers of separately sliced objects, but it would be hard to safeguard that there is enough space between objects.

### Affected users and/or printers

I think this would be awesome for everyone printing multiple objects that are too big for one-at-a-time printing.

### Additional information & file uploads

Would also be very nice for objects that consist of multiple pillars/parts, like this object that I'm currently printing:
[image]

GregValiant, 2024-12-05:
This has been requested at least once previously ([#14051](https://github.com/Ultimaker/Cura/issues/14051)).
The problem I see is that although it is possible for you or me to look at a model and determine that multiple layers could be printed in an area, there is no AI in Cura that can make the same sort of determination.

Occasionally I have noticed that Cura will print two layers on an island before moving to the second island and printing two layers.  What it is really doing is printing the second half of layer 10 on island 1, then moving up and printing the first half of layer 11 on island 1 before traveling to island 2. So it isn't really printing two layers, it's just optimizing travel.

x3e, 2024-12-09:
Thanks for your response, Greg! I'm not sure I fully understand the objection you raised. Let me try to clarify:

I don't think "AI" would be necessary here, as long as the number of layers to be printed "in advance" can be manually set in advance. For most printers, the nozzle is at least a few millimeters lower than the rest of the gantry, so it's possible to print several layers while ensuring there are no collisions.

What I envision is a setting (preferably in the "printers">"machine settings">"extruder N" settings) where I can specify two parameters:
- **Nozzle protrusion** – the vertical distance (in millimeters) between the lowest point of the nozzle and the lowest point of the gantry.
- **Nozzle body diameter** – the outer diameter of the nozzle, which could serve as the minimum spacing between "islands."

The nozzle protrusion be used to calculate the maximum number of layers that can be printed before moving to the next object or "island," while the nozzle body diameter can be used as the minimum spacing between objects/islands.

An implementation could work as follows:
1. Calculate M, where M=nozzle protrusion/layer height.
2. After slicing, analyze each layer. Use the "Nozzle body diameter" as the minimum spacing between print lines and group these lines into "islands."
3. If multiple islands exist in a layer X, look M layers ahead in the stack. For each subsequent layer, check if the islands:
   - Overlap with the islands on layer X.
   - Have minimal or no overhang compared to the islands on layer X.
4. If the islands on the next layers (X+1 to X+M) fully overlap and have little to no overhang, those layers can be printed immediately after the islands on layer X rather than waiting for the next layer in sequence.

GregValiant, 2024-12-09:
Thanks for that.  That's a nice, concise, description.
Some of the code for that may already be in place as it's the sort of thing that "One at a Time" determines.

HumbleDeer, 2024-12-15:
It would, however, limit the amount of layers grouped into one to a number that corresponds to the distance between the nozzle and the toolhead. So the amount of layers that would be printed at once would be fairly limited. If that's in line with the expected functioning, then I do think this could be useful.