Ultimaker/Cura #19996: Trying to slice a model I had to fix. Prusaslicer handles it fine

allanonmage, 2024-12-04:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Ender 3v2 that I had to set up manually

### Name abnormal settings

1mm nozzle

### Describe your model

I fixed it with an online tool.

### Add your .zip here ⬇️

[files.zip](https://github.com/user-attachments/files/18012547/files.zip)

Before and after fixing.  I'm using the fixed version in Cura.
[CE3E3V2_base_MSI_Version_fixed.zip](https://github.com/user-attachments/files/18012551/CE3E3V2_base_MSI_Version_fixed.zip)

allanonmage, 2024-12-04:
I've been having problems slicing this model, and I got through the first problem by adjusting some settings.  Since Cura kept nagging me to submit a bug report, I did.  I thought it would be more automated, but the directions are pretty good.

This is where I got the file:
https://www.thingiverse.com/thing:991811/

This is where I "fixed it":
https://www.formware.co/onlinestlrepair

Like I said above, Prusa Slicer 2.7.1 was able to work with the file fine.  I have multiple printers.

GregValiant, 2024-12-04:
Thanks for the report.
This is a duplicate of what was a deluge of reports regarding slicing errors in 5.5 and 5.6.  I'll mark this as a duplicate and go ahead and close it.  Most of the main fixes were done for 5.7.0 and 5.7.1 is very good.

Your only real option is to upgrade Cura.  I spend a lot of time with the current version (5.9.0) but I actually liked 5.8.1 better as 5.9.0 seems to be having issuers with the new Zseam settings.

I've never had a problem upgrading and I have a lot of Cura versions installed (10 out of my 22 installed slicers at last count).  In addition, my installation is heavily customized so I need to hand-copy some files when I upgrade.

You don't have to un-install 5.6.  I would create a project file and also export your custom profiles and materials.  Any other customizations (post-processors, machine definitions, etc.) should be safeguarded with backups just in case.

closed 2024-12-04 by GregValiant