Ultimaker/Cura #19997: Support

CORR-CONTROL, 2024-12-04:
### Is your feature request related to a problem?

Support är generellt för en hel utskrift, antingen totalt Support över allt eller ingen support alls. Men support tar tid och på många utskrifter finns inget behov av support på vissa platser, men på andra platser skall den var kvar. 

### Describe the solution you'd like

Jag vill att man skall kunna med musen markera var man inte vill ha support och då blir det ingen support på just denna plats, men på alla övriga platser blir det support som ju det nuvarande programmet väljer automatiskt. Där man med musen valt bort support ändras färgen så att man tydligt ser att här kommer ingen support att placeras.

### Describe alternatives you've considered

Support skall inte total om man väljer detta, utan skall kunna tas bort på de platser som man väljer manuellt med musen.

### Affected users and/or printers

Inga onödiga supportytor behöver skrivas ut.

GregValiant, 2024-12-05:
This is the exact function of Support Blockers.
Here the blocker is in place and covers a portion of the overhang and also covers the hole up above.
[image]

When slicing no supports are allowed to grow where the blocker touches an area that would normally require support.
[image]

Support blockers can be scaled and altered to do different things by using the Per Model settings.

Is that what you needed?

HumbleDeer, 2024-12-15:
I assume what they'd like is a feature similar to what I've heard Orca Slicer or Bambu slicer is commonly known to have, where you paint them on rather than using the blocky support blockers. I suspect or recall theirs also stop any support underneath if the feature that was requiring support is tagged as not needing support.

GregValiant, 2024-12-16:
"Paint On Supports" is available in PrusaSlicer, Orca, and Bambu.  They can shamelessly steal code from each other because they share a common base.  Cura is totally different under the hood.
I think this is on the To Do list for Cura but I don't know that for sure.

closed 2024-12-20 by github-actions[bot]