Ultimaker/Cura #19998: include a retract E-jerk value within the speed menu

iceman2511, 2024-12-05:
### Is your feature request related to a problem?

when using LA, it is required to have the E-jerk a high value of around 15-25 to get acceptable acceleration speeds out of a bowden printer for a marlin printer, but that also causes the extruder to be very audible when retracting for me at least, and reducing the e-jerk reduces or gets rid of the sound when retracting, so a seperate setting for retraction e jerk would benefit here.

### Describe the solution you'd like

just like i mentioned above, this setting would be in the speed section, and adjusting this E-jerk value would adjust the amount the retraction jerks, and only the retraction.

### Describe alternatives you've considered

no alternatives of what i can think of.

### Affected users and/or printers

everyone who uses a bowden printer, which has marlin on it, as well as linear advance turned on while printing.

GregValiant, 2024-12-05:
This would seem to be a personalization thing.
Go into your Startup Gcode and add a line at the beginning like "M205 E10".  That example will set the max jerk for the E stepper to 10.  You could also add an "M201 Exxx" to set the maximum E acceleration.
FYI The default Jerk on my bowden Ender 3 Pro is 5.00 and the Max E Accel is 1000.

I'll go ahead and close this as the proper way to handle this is to put it into the Startup for your particular machine.

iceman2511, 2024-12-05:
that completely aviods what im trying to do, the "M205 E10" sets the max e-jerk for the whole print, including linear advance, but i want a setting where it only adjusts the jerk setting for retractions, where it doesnt affect the print e-jerk or linear advance e-jerk.

GregValiant, 2024-12-05:
So what you are looking for is something like this?  This would cover retractions and the travel move and then reset the Jerk after the prime move.
M205 E5
G1 F3500 E-.8
travel
travel
G1 F3500 E.8
M205 E20

iceman2511, 2024-12-05:
yes, that looks exactly what i want to do.

GregValiant, 2024-12-05:
Then here ya go.  I have a Pull Request in for MaxVolumetricSpeed.  Adding Jerk was not difficult as all the other code was already there.
Unzip the file and place "MaxVolumetricSpeed.py" into the "scripts" sub-folder within your configuration folder.  It will be available with the other post processors under "Extensions".
This was a quick and dirty fix so right now you can run "Max Volumetric" with or without a Jerk adjustment, but to get a Jerk adjustment you must run Max Volumetric.  If you set the Max Volumetric Flow to a high number (like 75) then it won't have any effect.
[MaxVolumetricSpeed.zip](https://github.com/user-attachments/files/18029317/MaxVolumetricSpeed.zip)

iceman2511, 2024-12-05:
@GregValiant thank you! i will now test this out to make sure it works as expected.

iceman2511, 2024-12-05:
@GregValiant UPDATE: it doesnt work, i checked that my printer was silent, which it is, but i didnt check the speed of the printer, which it is printing at a very slow acceleration, i decided to look into the .gcode of the file it outputs, and it inputs the M205 X8 Y8 value, but after the travel and deretract, thhere is no M205 to convert it back to the original jerk setting.

GregValiant, 2024-12-05:
Post the gcode file.  It worked for me two minutes ago.
This is from a gcode I just sliced.  This is with "Jerk only" and no Volumetric.
M205 E5
G1 F1500 E2493.23983
M205 E20
G0 F7200 X115 Y109.368
G0 X66.172 Y109.44
G0 X66.1 Y109.934
M205 E5
G1 F1500 E2500.23983
M205 E20

With Volumetric it's 
M203 E25.0
M205 E5
G1 F1500 E150.74565
M203 E4.99
M205 E20

So that looks good to me.

iceman2511, 2024-12-06:
It's all working good now, I just didn't follow the instructions within the script, having the print jerk lower than it should be.

GregValiant, 2024-12-06:
There isn't a separate setting for the "E-Jerk" so I picked the "Print Jerk" for the reset value.  You can set that to whatever you need and then make other adjustments with the "per feature" jerk settings.
I could have used "Default Filament Jerk" from the Machine Settings as the reset, but that isn't what people always use and it really has no effect within Cura.  Print Jerk seemed like a better choice.

If the pull request is accepted I'll come back and close this.