Ultimaker/Cura #19999: Error 2171 when printing any slice from 5.9 on a creality ender 3 V3 SE (with nebula pad)

dreamkiller52, 2024-12-05:
### Cura Version

5.9

### Operating System

Windows 10

### Printer

Creality Ender 3 v3 se with nebula pad

### Reproduction steps

slice any file in Cura 5.9.0 (works with a 5.8 slice)
copy to usb key
lauch on the printer , 
when the nozzle is at the desire temperature the head move in the center on the plate, do a Z test , move to the front left corner and then the printer throw an 2171 exception and the head stay here

### Actual results

the printer throw an 2171 exception

### Expected results

the printer should print something (line purge)

### Add your .zip and screenshots here ⬇️

[project_gcode.zip](https://github.com/user-attachments/files/18027374/project_gcode.zip)

GregValiant, 2024-12-05:
Thanks for the report.
This is kind of weird.
This line is in the StartUp Gcode section of the 5.9 gcode file you posted.
M109 S[material_print_temperature_layer_0]
Because the keyword is enclosed in square brackets it didn't get replaced.  My guess here is that the "2171" error is from the Nebula pad and is "command not understood" and that is causing the Nebula pad to hang.

The weird part is that neither of the project files you posted have square brackets in the line, and the Cura Ender 3v3se definition file does not have square brackets either.  PrusaSlicer and it's clones use square brackets.  Cura uses curly brackets.

Did you happen to open a project file from somewhere else?

I'm removing the bug label and I'll close this as it does not appear to be a Cura issue.  Go into the Cura Machine Settings and check your "StartUp Gcode" and make sure that all the lines with keywords have curly brackets.

closed 2024-12-05 by GregValiant

dreamkiller52, 2024-12-05:
I'll have toncheck that tommorow.
I try with a benchy and have the same error

When i install 5.9 i have a strange behavior , temp of the material and of the bed didn't follow the settings set in material management until the computer reboot.

dreamkiller52, 2024-12-06:
effectively my Start gCode is not the same in 5.8 and 5.9 where does it come from ?

GregValiant, 2024-12-06:
The StartUp Gcode comes from the printer definition file.
It can be changed manually by the user, and it can be over-written if you open a 3mf project file.  (A 3mf model file doesn't have any printer information in it and will leave your installed printer alone.).

Post processor could also change it.  I am familiar with ChangeAtZ and it would not have done it.  The Temperature Tower post-processor you used doesn't seem to be a UM Cura package.

I checked the Creality definition files for your printer (that are included with UM Cura) are they are correct in both 5.8 and 5.9.  Even that project file you posted was correct.  The only thing I saw wrong was the line in the Gcode.  I have no idea how that could have happened, but it wasn't Cura.

If there is more than one keyword in the StartUp or Ending Gcodes, and if even one of them is wrong, then the substitution of numbers for ALL the keywords fails.  In your case it failed because Cura doesn't understand square brackets.

I would install another Ender 3v3se in Cura, and then delete your current printer.  The StartUp gcode I see in that 5.9.0 gcode did not come from UM Cura.
This is what is in the definition file, and it is identical to BOTH of the project files you posted.
```
M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate

G28 ;Home

M420 S1; Use saved mesh leveling data

G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
G1 X-3 Y20 Z0.28 F5000.0 ;Move to start position
M190 S{material_bed_temperature_layer_0} ; Set bed temperature and wait
M109 S{material_print_temperature_layer_0} ; Set hotend temperature and wait
G1 X-3 Y100.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X-2 Y100.0 Z0.28 F5000.0 ;Move to side a little
G1 X-2 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0 ;Reset Extruder
G1 E-1.0000 F1800 ;Retract a bit
G1 Z2.0 F3000 ;Move Z Axis up
G1 E0.0000 F1800 
```
This is from the wonky gcode file that was allegedly sliced in 5.9.0.
```
M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate
G28 ;Home
M420 S1; Enable mesh leveling
G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position
M109 S[material_print_temperature_layer_0]
G1 X10.1 Y145.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X10.4 Y145.0 Z0.28 F5000.0 ;Move to side a little
G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0  ;Reset Extruder
G1 E-1.0000 F1800 ;Retract a bit
G1 Z2.0 F3000 ;Move Z Axis up
G1 E0.0000 F1800
```
It is missing a couple of blank lines.  The purge lines have moved.  It is missing the bed temperature line, and the hot end line has square brackets instead of curly brackets.

dreamkiller52, 2024-12-06:
i just delete and re add the printer in 5.9 always the square braquet and different printer definition 
5.8 GCODE 
`M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate

G28 ;Home

M420 S1; Use saved mesh leveling data

G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
G1 X-3 Y20 Z0.28 F5000.0 ;Move to start position
M190 S{material_bed_temperature_layer_0} ; Set bed temperature and wait
M109 S{material_print_temperature_layer_0} ; Set hotend temperature and wait
G1 X-3 Y100.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X-2 Y100.0 Z0.28 F5000.0 ;Move to side a little
G1 X-2 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0 ;Reset Extruder
G1 E-1.0000 F1800 ;Retract a bit
G1 Z2.0 F3000 ;Move Z Axis up
G1 E0.0000 F1800 
`

5.9 GCODE
`M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate

G28 ;Home

M420 S1; Enable mesh leveling

G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position
M109 S[material_print_temperature_layer_0]
G1 X10.1 Y145.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X10.4 Y145.0 Z0.28 F5000.0 ;Move to side a little
G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0  ;Reset Extruder
G1 E-1.0000 F1800 ;Retract a bit
G1 Z2.0 F3000 ;Move Z Axis up
G1 E0.0000 F1800 `

I remember i install manualy a printer definition file in an older version of cura 5.7.x when the Ender3 v3 se wasn't reconized maybe it's a file remaining somewhere

GregValiant, 2024-12-06:
I wonder if there is a bad definition file in your Configuration Folder.  You can use the "Help | Show Configuration Folder" command to locate it and then look for the Definitions folder.  Then shut Cura down.
Within that Config Definitions Folder look for "creality_ender3v3se.def.json" and delete every file with that name.
Then go to the "Definition Changes" folder and look for any file that starts with "creality_ender3v3se" but not "extruder" in them.  Delete those non-extruder files.
When you restart Cura it should be forced to use the regular definition file.  If there is still a problem then un-install Cura 5.9.0 and then re-install it.  The installation files are [HERE](https://github.com/Ultimaker/Cura/releases) under "Assets".

What you are seeing is not a corruption but an intentional change that someone made to the file.