Ultimaker/Cura #20000: Top surface line directions dictated by odd/even layer position not settings

PartySausage, 2024-12-05:
### Cura Version

5.9.0

### Operating System

Windows 10

### Printer

Ender 3 S1 Pro

### Reproduction steps

I did a test design to get my head round why the line directions on the top surfaces of a model were different depending on the layer where surface was in the model, 
The test model was a series of steps that had top surfaces on both odd & even layers & I tested it with different top layer/skin layer combo's. The Top/Bottom & Skin Line Directions were both set to [90, 0].

### Actual results

The result was that it doesn't matter what combination you use the line direction is determined whether the surface falls on an odd/even layer, which seems to be the wrong way round to me

### Expected results

 I would have thought the line direction on the top surface should be orientated according to say the 5th layer based on the configured line directions for 5 layers & work down from there with the top most layer being in the same orientation regardless of the layer it falls on. With the way it works now you're pretty much taking pot luck with the line directions in a sliced design based on the layer not where the top layer sits in the slice with the only way to ensure that all the top surfaces lines align the same way is to set all the line directions the same eg [90, 90, 90, 90, 90] for the top/bottom & skin line directions... Not exactly ideal for a robust surface

### Add your .zip and screenshots here ⬇️

[image]
[image]

[Top Layers Test.zip](https://github.com/user-attachments/files/18029600/Top.Layers.Test.zip)

GregValiant, 2024-12-06:
My two pennies worth...
It acts like it starts with the first angle in the Angle List on the bottom layer and indexes through the angle list for each layer.  Whatever angle is current is the one used.
With two angles then it's every other layer.  With 3 angles it's every 3rd layer.  So it repeats "1st angle" every "length-of-list" layer.

PartySausage, 2024-12-06:
Thanks for the explanation @GregValiant, pennies even a couple can be gold. Between you & @Slashee I've got a good understanding of the why & how it works now. With my question hence the bug report being should it work that way? 
The top/bottom & skin line directions in my opinion shouldn't be determined by the layer no. where they appear & should be independent based on their settings when even just minutely scaling the model can change how the lines are orientated.
Only the combined brains behind the dev can decide whether the current operation is how it's intended to work or if my suggestion is a worthy change & I'll just work with the result