Ultimaker/Cura #20005: no infill

Coffeewithkids, 2024-12-06:
### Cura Version

5.4.0

### Operating System

windows 

### Printer

ender3

### Reproduction steps

stander profile

### Actual results

no infill

### Expected results

no infill

### Add your .zip and screenshots here ⬇️

none

Coffeewithkids, 2024-12-06:
pleas  help me

Asterchades, 2024-12-08:
We can't with how little information you've given. For all anyone can tell you've set the _Infill Density_ to 0% so it's not generating infill because you've told it not to.

This is why the form you submitted requested a Project File, screenshots, or something that might otherwise be useful. Without any information to go by beyond "no infill" it's impossible to perform any diagnostics at all as there are plenty of expected reasons as to why that might be the cases.

If you can provide some information, someone should be able to assist you.

closed 2024-12-22 by github-actions[bot]