Ultimaker/Cura #20008: Print name appears double (.gcode and .ufp)

spanzetta, 2024-12-07:
### Cura Version

5.9.0

### Operating System

Windows 10

### Printer

Ender 3 Pro

### Reproduction steps

1. Slice any 3D file and send it to Octoprint
2. See the name in the popup that appears after have sent the file to Octoprint
See screenshot below

### Actual results

the name appears twice..  as **name**.gode**name**.ufp (see screenshot)

### Expected results

show the name only once (.gode or .ufp)

### Add your .zip and screenshots here ⬇️

[image]

fieldOfView, 2024-12-07:
This is an issue in a 3rd party plugin and should be reported here instead:
https://github.com/fieldofview/cura-octoprintplugin/issues

You can just copy/paste this the issue text in the new issue form there.

spanzetta, 2024-12-07:
ok thanks

closed 2024-12-07 by spanzetta