Ultimaker/Cura #20012: Problem with sliced objects.

Svartjanne, 2024-12-08:
### Cura Version

5.9.0

### Operating System

Windows 11

### Printer

Creality CR10-s Pro

### Reproduction steps

1. I downloaded an STL-file from Thingiverse and sliced it in cura. After slicing i saved the G-code file to an SD-card to put in the printer. The printer does not find the file and when inserting the SD-card to the computer and opening the G-code file in Cura again the Program shows: " No estimated time available " & " Print setup disabled. G-code file can not be modified ". I´ve hade these problems before and i can´t explain them. It is like the G-code file is empty.
2. The printer is working as it should and i could find an old saved file on the SD-card that i printed a few days ago.
3. I formated the SD-card but the problem persists.
4. I´ve tested changing some settings in Cura, both printer settings and format setting without any success.
5. I have also tried opening the STL-files with Fusion and then exporting them to G-code but nothing works.
6. Uninstalling and installing Cura again have been made but no difference.
7. The particular STL-file is not the problem either since i´ve tried to open other models also.

### Actual results

Absolutely nothing changes when i have tried the above mensioned things.

I´m thinking there is a software problem maybe between a windows update and the new Cura version since it appeared out of nothing and the program worked a few days ago.

### Expected results

The G-code file should be possible to open and make changes as usual when opening the G-code file in Cura.
The printer should also be able to find the G-code file. As mentioned it seems like the G-code file is empty but there´s a model in Preview only.

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2024-12-08:
Thanks for the report.
Your image shows the gcode file previewed in Cura.  That means that the Gcode file is not empty.

A gcode file is a simple text file and does not contain a model.  Cura is not a text editor and so shows that "information" message whenever you open a gcode file because the gcode "text" cannot be modified from within Cura.
If you need to change the way you sliced then you must bring the model back into Cura, change your settings, slice again, and re-save the file as a gcode.

I understand you might be having a problem with the printer recognizing the gcode file but that is not a Cura problem.
I'll remove the bug label and go ahead and close this.  I hope you understand.

closed 2024-12-08 by GregValiant

Svartjanne, 2024-12-09:
Hey Greg,

Any suggestions of what i can do to make it work again?

The only difference that made it ” not work ” is a Windows update i did last friday.

Svartjanne, 2024-12-09:
I have had this problem with Cura version 5.8.0 a few times but it disapeared when updating to 5.9.0 but now it is back again.

GregValiant, 2024-12-09:
That indicates it is a local problem rather than an actual glitch in the Cura code.  If you are able to print the "stock" gcode files from the SD, then hopefully it isn't a card reader problem.
Slice something simple (like a calibration cube) and save it to the SD card.  Then copy that file back to your hard drive.  Zip the gcode file that you copied back and post it here.  If there is some sort of corruption in the SD file it will show up.
My printer is an Ender 3 Pro and it's pretty similar to yours so I have some familiarity.  I'll take a look at the code.

Every once-in-a-while a problem like this gets reported.  It's never been a Cura problem but rather something going on with the printer.
As a check - you can try printing something over the USB.  "USB Printing" can be glitchy and it isn't advisable to use it for everything but as a check it may give a clue.

Svartjanne, 2024-12-09:
Ah alright,

Here you go:

[CCR10SPRO_xyzCalibration_cube.zip](https://github.com/user-attachments/files/18064295/CCR10SPRO_xyzCalibration_cube.zip)

Thank you Greg.

Svartjanne, 2024-12-09:
I downloaded the Cura 5.8.1 and now it is working.

Cura looks the same but i can find the file on the sd card when i put it in the printer now.

Thanks for your support.

GregValiant, 2024-12-09:
That Gcode file is fine and should print.

A little history...
When the original CR-10s came out they had a 16gb limit on the SD card size.  They would accept a larger card but couldn't read them unless they were partitioned into smaller chunks.
The CR-10s were also limited to 16 character file names.  "My Print is cool.gcode" would work but "My print is really cool.gcode" would not.  Files with long names would not show up on the display and could not be selected on the LCD.

Svartjanne, 2024-12-09:
Alright! :)

Yeah, iv'e encountered the too long name problem i think. But the Sd card problem i were not aware off.

Maybe they are uppdated on the Pro? I just put a 32gb card in thinking maybe the 16gb was failing and the 32gb worked fine after installing the 5.8.1 version. The 16gb also works when putting that one in now.

GregValiant, 2024-12-09:
This goes along with one of my sayings "Computers are not an exact science."
There is very little difference between 5.8.1 and 5.9.0.  There weren't any changes that would effect the actual gcode and whether or not the printer found it acceptable.