Ultimaker/Cura #20014: Rejected code when using python in Start G-code it is also extruder_nr.

toreandreas, 2024-12-09:
### Cura Version

5.9.0

### Operating System

Windows 10

### Printer

Ender5 - Klipper

### Reproduction steps

In Cura Start G-code:
```
{if "PLA" in material_type or 2 == extruders_enabled_count}
M207 T1 S{retraction_amount,1} F{retraction_retract_speed,1} Z{retraction_hop,1}
{endif}
```

### Actual results

`M207 T1 S4.0`

### Expected results

`M207 T1 S4.0 F60 Z0.1`

### Add your .zip and screenshots here ⬇️

The "old" method also doesn't work:
`{"" if "TPU" in material_type or 1 == wall_extruder_nr else ";"}M207 T1 S{retraction_amount,1} F{retraction_retract_speed,1} Z{retraction_hop,1}`

GregValiant, 2024-12-09:
Thanks for the report.
I can duplicate this.

It appears that only the first "replacement" on the line is added.  Any additional parameters and replacements on the line seem to be ignored ... BUT ... if there is a comment at the end it does gets added.

You can fool that by adding the parameters one line at a time.
```
{if "PLA" in material_type or 2 == extruders_enabled_count}
M207 T1 S{retraction_amount,1}
M207 T1 F{retraction_retract_speed*60,1}
M207 T1 Z{retraction_hop,1}
{endif}
```
Someone from the Cura team (@wawanbreton ??) will take a look.

HellAholic, 2024-12-10:
Added a ticket to the backlog, 
Might be changed to a feature/feature gap but just for reference so it's linked.
Internal Cura reference: CURA-12358