Ultimaker/Cura #20018: An option to allow firmware to also process Z Hop, when Firmware Retraction is enabled.

toreandreas, 2024-12-09:
### Is your feature request related to a problem?

Briefly: Marlin also has the ability to change Z Hop, when Firmware Retraction is enabled. Cura does not use this ability.

Instead of trying to explain the problem (which I am not good at) again, I will refer you to this [thread.](https://community.ultimaker.com/topic/45386-testing-with-firmware-retraction/)

### Describe the solution you'd like

A setting under Printer Settings > Firmware Retraction:
Firmware Retraction Z Hop.
When enabled, Cura hands over Z Hop to the printer.

### Describe alternatives you've considered

In the same [thread](https://community.ultimaker.com/topic/45386-testing-with-firmware-retraction/) as above, a kind of solution is shown.

### Affected users and/or printers

Anyone who wants to test Firmare Retraction with Z Hop.

GregValiant, 2024-12-09:
You can do this using ["replacement patterns"](https://files.fieldofview.com/cura/Replacement_Patterns.html) in your StartUp Gcode.

```
M207 F{retraction_retract_speed*60} S{retraction_amount} Z{retraction_hop}
M208 F{retraction_prime_speed*60}
```

See if that works for you.

toreandreas, 2024-12-09:
Your code works as is. This line is not needed:
Z{retraction_hop if machine_endstop_positive_direction_z else 0.0}

As long as "Z Hop When Retracted" is disabled, everything works as it should.
What I'm thinking is that maybe others besides me might need this feature!
I think it's a good thing to be able to fine-tune Z Hop as well.

GregValiant, 2024-12-09:
Remember that UltiMaker does the heavy lifting for Cura and they manufacture a line of printers.  They are not going to pay their people to produce these changes for a 3rd party printer that was customized by a 4th party and is running 5th party firmware.

So it is necessary to get the tools that are available to work for you (workarounds), or learn to write your own custom code to do what you need.  That is how I got started writing post-processors.

GregValiant, 2024-12-12:
Unzip this file and place the "ConfigureFirmwareRetractPlugin" folder into the "plugins" sub-folder that is in your "Configuration Folder".  (Yes, there is a double folder and both have the same name.  That is true for all plugins.)

The plugin adds a new setting "Configure G10/G11" under "Firmware Retraction" in the machine settings.  If you enable that, then "Enable ZHops (T0)" will be available (and for a dual extruder printer, "Enable Zhops (T1)" will be also available.)
The settings in M207 and M208 come from the Travel settings and must be configured there.  You have to "enable z-hop on retraction" in the Travel settings to set the z-hop height.  Once you have it set then "Z-hop on retraction" can be disabled again.
The lines are inserted just before the startup in the gcode file like this (for a dual extruder machine).
Also for a dual extruder printer; the "Tool Switch Amount", "Tool Switch Extra Prime" and "Tool Switch Retract Speed" come from the Dual Extrusion settings and would need to be altered there.
```
;Generated with Cura_SteamEngine 5.9.0
;
M207 T0 S7 F1500 W16 Z0.2       ; Configure G10 Retract ('T'tool num, 'S'retract amt, 'F'retract speed, 'W'tool switch amt, 'Z'Z-hop hgt)
M208 T0 S0 F1500 W0 R1200       ; Configure G11 Prime ('T'tool num, 'S'extra prime amt, 'F'prime speed, 'W'tool switch extra prime amt, 'R'tool switch retract speed)
M207 T1 S7.0 F1500 W16 Z0.2     ; Configure G10 Retract
M208 T1 S0 F1500 W0 R1200       ; Configure G11 Prime
;
```

[ConfigureFirmwareRetractPlugin.zip](https://github.com/user-attachments/files/18113022/ConfigureFirmwareRetractPlugin.zip)
Here are the settings:

[image]

closed 2024-12-12 by GregValiant

GregValiant, 2024-12-12:
I see a possible problem at the start of all layers.
LAYER:0
M106 S0
M204 S500
M205 X8 Y8
G0 F10500 X171.351 Y44.311 Z0.25  >>>> this drops the Z to "Initial Layer Height"
;TYPE:SKIRT
G11 >>>>>>>>>>>>> If there is a "Hop down" in G11 then it may put the nozzle at a negative Z height.

the following layer:
G10         >>>>>> Retract with a hop up
;MESH:NONMESH
G0 F600 X96.596 Y185.762 Z0.5  >>>> Cura sets the height of the next layer
;TIME_ELAPSED:374.433529
;LAYER:1
;TYPE:SUPPORT
G11   >>>>>>>>>>>>> If there is a hop down here then the nozzle would be too low.

It would appear that the other parameters can be adjusted, but maybe not Z-hops even if they are turned off in Cura Travel Settings.

toreandreas, 2024-12-12:
Hi. I see you've been here, @GregValiant . I've been to a funeral today. (No one in the immediate family.) So maybe I'll look at this later tonight, but probably tomorrow.

toreandreas, 2024-12-12:
First impression, great!!!
About the problem with "the start of all layers". I have a solution for this. I have a routine in Klipper that resets all Firmware Retraction. This is started with a line in Start G code. But I can't see that e.g. Marlin has a code for this. You have to remember to reset all extruders, not just the active one.

I switch extruder no. 2 between TPU (direct drive) and PETG (Bowden Tube). So physically switch extruder and hotend.

Before I used Cura as if I had 2 Ender 5 printers. The "first" configured with the first extruder PLA, and the second for PETG. The "second" was configured with the first extruder PLA, and the second for TPU.

Now I have used Cura as if I have 1 Ender 5 printer, with 3 extruders. PLA, TPU and PETG respectively. Then of course TPU and PETG can't be active at the same time. And then I have a post script that searches for all occurrences of T2, and changes them to T1.

Because of your plugin I have to change back. But that's okay. You win something and you lose something.

In my Klipper macro files I have not included the parameter for switch (M207/M208 W) (G10/G11 S1). So there I have now got a job. It may take some time. But there are some holidays coming up soon, so hopefully not for too long.

GregValiant, 2024-12-13:
I removed my request for the plugin to be added to Cura because of the "Z" effect and the very real possibility it would be disastrous.
The Cura "Z" hops work fine and are adjustable as to how much, and when.  M207 and M208 can be adjusted in the StartUp Gcode which is easier anyway.

toreandreas, 2024-12-13:
> I removed my request for the plugin to be added to Cura because of the "Z" effect and the very real possibility it would be disastrous. The Cura "Z" hops work fine and are adjustable as to how much, and when. M207 and M208 can be adjusted in the StartUp Gcode which is easier anyway.

Ok.
I'm not sure how the R parameter in M208 is used. I don't see Marlin and Reprap using this. "'R'tool switch retract speed" suggests that it is part of M207 retract and not M208 recover (prime). It is taken from 'Nozzle Switch Retraction Speed', not 'Nozzle Switch Retract Speed'.

GregValiant, 2024-12-13:
I took the parameters from the Marlin site.
"M208 [F<feedrate>] [R<feedrate>] [S<length>] [W<length>]
Parameters
[F<feedrate>]	
Recover feedrate (units/min)

[R<feedrate>]	
Swap recover feedrate (units/min)

[S<length>]	
Additional recover length. Can be negative to reduce recover length.

[W<length>]	
Additional recover swap length. Can be negative to reduce the length.

toreandreas, 2024-12-14:
> I took the parameters from the Marlin site. "M208 [F] [R] [S] [W] Parameters [F] Recover feedrate (units/min)
> 
> [R] Swap recover feedrate (units/min)
> 
> [S] Additional recover length. Can be negative to reduce recover length.
> 
> [W] Additional recover swap length. Can be negative to reduce the length.

Of course. Don't know how I managed to miss this. It's just embarrassing.