Ultimaker/Cura #20019: Z seam alignment "random" concentrates on back side

upatamby, 2024-12-09:
### Cura Version

5.9.0

### Operating System

Windows 10

### Printer

Ultimaker 3E

### Reproduction steps

Load a cilindric model. Choose Print Settings and select Walls, Z Seam Alignment = Random. Choose Preview and select Color Scheme, Line Type and check Starts.
Slice the model.

### Actual results

After slicing is done you can see that the starts of the layers seem to be random, but they are all concentrated at the back side of the model. 

### Expected results

I would expect that the starts are more or less equally spread over the hole surface of the cilinder

### Add your .zip and screenshots here ⬇️

[image]
[issue.curaproject.3mf.zip](https://github.com/user-attachments/files/18068605/issue.curaproject.3mf.zip)

BogusF, 2025-01-04:
It is a [known issue](https://github.com/Ultimaker/Cura/issues/8464), feel free to chime into the discussion, maybe if there's enough feedback, they start doing something about it, instead of deflecting.

willstoeckel, 2025-03-06:
+1 for this problem. Very annoying and makes concentric collapsible prints difficult

[image]