Ultimaker/Cura #20022: Reset all model position has problem with more then 2 objects

kszitas, 2024-12-10:
### Cura Version

5.9.0

### Operating System

Windows 11

### Printer

Method X

### Reproduction steps

Load an 3mf file with 3 objects.
Select all models
Reset all model positions.

### Actual results

2 objects are moved to their positions as were definied in 3mf model.
The 3. object is at the same position as the 2nd one.

### Expected results

should all positions be as in the 3mf file definied.

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2024-12-11:
Thanks for the report.
I can duplicate this.  Even the first load isn't right.

Someone from the Cura team will take a look.
The models in this 3mf were located at:
1. Cube at X0 Y0 Z0
2. Pyramid at X-75 Y0 Z0
3. Cylinder at X0 Y-80 Z0

Image from 3D Builder:
[image]

Opening the 3mf file in Cura:
[image]

Clear the build plate and open it again.
[image]

[New Scene.zip](https://github.com/user-attachments/files/18089148/New.Scene.zip)

HellAholic, 2024-12-12:
I think there is an issue open for this one on the backlog, but with more reports I can bump it higher.
Until the bumping is successful, after the load press Ctrl + Z, and that puts the model back in the correct relation.
The issue is that it tries to load the 3mf in the middle of the build plate but it does not have the reference to the coordinate of the buildplate (machine) so just loads individual models in center and does an arrange. While what it should do is load the entire model group and do a center view on the bounding box of the entire group.
(might be mixing some terminology but that's the gist of things).