Ultimaker/Cura #20026: "Drop to build plate" drops under the build plate, not over

kszitas, 2024-12-12:
### Cura Version

5.9.0

### Operating System

Windows 11

### Printer

Method X

### Reproduction steps

1. Export a Body as Mesh from Fusion 360 in 3mf format
2. Import model to Cura
3. Drop to build plate

### Actual results

Drop to build plate drops it under the Build Plate

### Expected results

Should be over the plate, not under

### Add your .zip and screenshots here ⬇️

[image]
[TestFile.zip](https://github.com/user-attachments/files/18112539/TestFile.zip)

GregValiant, 2024-12-12:
Thanks for the report.
There appears to be something going on with Cura and 3mf files but I can't tell if this is a part of that situation.  #20022 reported improper loading of 3mf's.

Cura should use the location data that is part of a 3mf file and bring the model in at that location.  That is the expected behavior.
When "Automatically drop models to build plate" is checked, I would think that the model should come in at the stored XY but that the Z should be adjusted to put the model at Z=0.  That seems to be the case when the model XY puts it over the build surface in Cura, but fails when the XY is off the build plate (like it is in your test file).  This is a question of semantics but since the setting is "Drop models..." and not "Raise models..." it might be doing exactly what the code tells it to since the virtual build plate in Cura is finite in size.

Someone from the Cura team will take a look.

kszitas, 2024-12-12:
" Z should be adjusted to put the model at Z=0" I think should find the model with the biggest volumen inside the 3mf file, and that is moved to Z=0, all other models relative to the biggest model's new position

GregValiant, 2024-12-12:
The models are within a "bounding box" and that is what is used for positioning.  It is the bottom of that "bounding box" that is set down to the build plate.  The lowest model within the box determines where Z=0 will be.  It has nothing to do with the size of any individual model.

If the model location in the 3mf is below Z=0, then it will not adjust upward to the build plate.
If I drag any model (3mf, STL, OBJ, etc.) below the build plate, it never "jumps up" to rest on the build plate.  It will only "Drop to the build plate" which is exactly what the setting and tooltip say will happen.

[image]

I don't really see a bug here as it is doing what it says it will do.  Maybe this should be a feature request?
"Automatically drop models to build plate" does seem to be affecting the XY locations and that is what the other bug report is about.  I don't think that should be happening.