Ultimaker/Cura #20029: missing build plate temperature option

Jimmyracle, 2024-12-13:
### Cura Version

5.9.0

### Operating System

mac os

### Printer

creality k1 max

### Reproduction steps

i looked for the build plate temperature adjustment option. i focused on the materials pulldown menu to find this option. couldn't locate it. checked other pulldown menus, still could not locate it. went under preferences to see if the build plate temperature setting is selected, it is selected.

### Actual results

cannot change build plate temperature, due to missing option from materials pulldown menu.

### Expected results

with the option present, the temperature of the build plate should be adjustable.

### Add your .zip and screenshots here ⬇️

[image]

HellAholic, 2024-12-13:
Try the search bar in either the setting visibility or the print setting
[image]

Jimmyracle, 2024-12-14:
[image]
 The first screenshot was a mistake.

HellAholic, 2024-12-14:
Ok in that case, since you have a little info button next to the build plate temp, I am guessing that in the machine settings the heated bed is unchecked. Which is what the tool-tip is probably say if you mouse over the info button next to the build plate temperature in the setting visibility you shared.
So in the preferences go to the printers, select your K1 Max, select machine setting, and check if the heated bed is checked.
[image]

If it is checked. select your printer > go to help > export package for technical support, this creates a 3mf file with the machine and your logs, make a zip out of that and drop it in the report here.

Jimmyracle, 2024-12-14:
You solved my problem, thank you HellAholic.
[image]

closed 2024-12-14 by Jimmyracle