Ultimaker/Cura #20033: Cura hangs up forever when opening. Seems like waiting for backup to initiate

HankLumbard, 2024-12-14:
### Cura Version

5.8.1

### Operating System

Mac OS 15.1.1

### Printer

Reality ender 3 v2 

### Reproduction steps

Just opening the program. Can't load models to slice for 5 minutes

### Actual results

Program responds to everything except loading a model

### Expected results

should be able to add str file and have it show up

### Add your .zip and screenshots here ⬇️

have nothing to show. It's been like this for maybe a month?
[popv1.3mf.zip](https://github.com/user-attachments/files/18137578/popv1.3mf.zip)

HellAholic, 2024-12-14:
Sounds like a USB printing issue, If you're not using the USB printing function, disabling it will most likely resolve the issue.

Open the Marketplace > Click on the Manage packages > Scroll down to USB Printing > Disable USB printing and restart Cura
[image]

HankLumbard, 2024-12-14:
I should have added. I'm running octoprint and connected through the USB

Asterchades, 2024-12-15:
The USB plugin will only affect things if you're using Cura itself to host the print via USB. Otherwise OctoPrint has its own USB communication implementation which it handles itself, after Cura communicates with it via the OctoPrint plugin you would have installed.

madhead, 2024-12-15:
> Sounds like a USB printing issue, If you're not using the USB printing function, disabling it will most likely resolve the issue.
> 
> Open the Marketplace > Click on the Manage packages > Scroll down to USB Printing > Disable USB printing and restart Cura

That was it for me! Thanks!

P.S. Why did it work previously? I never ever used USB to communicate with my printer, and it was never a problem.

HankLumbard, 2024-12-15:
That did it for me! Madhead, same Q here. Never was a problem before.

HankLumbard, 2024-12-17:
I'm relatively new to GitHub. Shouldn't this stay open since there's clearly some kind of bug?

fieldOfView, 2024-12-17:
If this were the only report on the USB printing plugin slowing down Cura, it would remain open. As is, it is one of many issues that describe the same thing, so we close this one.

closed 2024-12-17 by fieldOfView