Ultimaker/Cura #20034: Not slicing porperly

plgarcia, 2024-12-15:
### Cura Version

5.9.0

### Operating System

Window 10 

### Printer

FLSun V400

### Reproduction steps

Th model does not slice properly. I check the model with blender and with meshmixer it is manifold.

### Actual results

It slices a cylinder including the object but not the object itself.
[image]
[image]

### Expected results

Slice te object

### Add your .zip and screenshots here ⬇️

[FV_supportEclairage-socle.zip](https://github.com/user-attachments/files/18140365/FV_supportEclairage-socle.zip)

plgarcia, 2024-12-15:
Note that CURA 5.8.1 slices this model accurately.

GregValiant, 2024-12-15:
Thanks for the report.
Cura is slicing this in 5.9.0 exactly as you have asked it to.  If you want the hole in the middle you need to turn off "Remove All Holes" in the Mesh Fixes settings.

I'll go ahead and remove the bug label and close this as it is a setting issue.  I hope you understand.

closed 2024-12-15 by GregValiant

plgarcia, 2024-12-15:
Sorry for that I did not even think I had selected this option. I believe that when I did it, it was thinking fix topology problems, but not changing the geometry. Thank you! it works fine.

GregValiant, 2024-12-16:
So long as it is working for you it's all good.
The other setting that is only useful in special situations is "Make overhangs printable".  That one puts a chamfer on the bottom of overhangs.
Both of them need to be turned off right after they are used or you sit there scratching your head wondering what the heck is going on.