Ultimaker/Cura #20035: Missing Dual Extrusion On / Off Check Box

Firedogs2x, 2024-12-15:
### Cura Version

5.9.0

### Operating System

Windows 10

### Printer

Modix

### Reproduction steps

I have settings visibility set to ALL.
I only have only Head One turned on.
My filament profile was set for to only print one material.
I have tried shutting down CURA and restarting it. No changes.

### Actual results

When I expand the Dual Extrusion category there is no longer the check box for turning on or off dual extrusions.
Even when I do turn on Head Two there still is no check box. 

### Expected results

In the past there was always a check box to turn on or off Dual Extrusion.

### Add your .zip and screenshots here ⬇️

[image]
[image]
[Blank Dual Extrusion Project 12-15-2024.zip](https://github.com/user-attachments/files/18141103/Blank.Dual.Extrusion.Project.12-15-2024.zip)

GregValiant, 2024-12-16:
Thanks for the report.
I don't recall a command to "Enable Dual Extrusion".  There are settings under "Settings | Extruder 1(and 2)" that allow you to enable or disable a particular extruder.
If you load the "SidebarGUI" plugin from the MarketPlace it adds checkboxes for enabling and disabling either extruder.
This with SidebarGUI loaded:
[image]

The arrow to the right of "Extruder 2" will hide or show the checkboxes, nozzle size, and material.

I don't think this is a bug so I'll remove the label.  Basic Cura is set up to access the extruders through the menubar and does not have checkboxes for the extruders.
See if loading the plugin has the effect you expected.

Firedogs2x, 2024-12-16:
Hello, Thank you for the prompt reply.
After relooking at Cura you are correct.
It seems I had what could be called a senior moment.
Somehow I thought the check box to turn on or off supports was suppose to be under Dual Extrusion.
Again thank you for your help.

closed 2024-12-16 by Firedogs2x