Ultimaker/Cura #20036: ChangeAtZ 5.3.0 crashing Cura when using layer nº

FabioM3, 2024-12-16:
### Cura Version

5.7-5.9

### Operating System

Windows 10 64bit

### Printer

Ender S1

### Reproduction steps

- Add an model
- Add script ChangeAtZ 5.3.0
- Change parameter to layer number
- Change the layer and/or other parameters

### Actual results

Cura crashes and closes.

### Expected results

Not crashing!

### Add your .zip and screenshots here ⬇️

[cura.zip](https://github.com/user-attachments/files/18143015/cura.zip)

GregValiant, 2024-12-16:
Thanks for the report.
I am familiar with ChangeAtZ and I've never had a problem with it.  There isn't anything in it that can cause a crash.  It just gets the Gcode from Cura, makes your changes, and sends the Gcode back.  It has only undergone minor changes in the last several years.
I looked through the log file and there is no record of a crash.
So I am unable to duplicate this with the information you provided.

Import a simple model and set Cura up in the manner that caused the crash.  Force the crash and then go to the Configuration Folder and zip the "cura.log" file and post the zip folder here.  A project file might be useful as well ("File | Save Project") that will need to be done before the crash.  The 3mf project file would also need to be zipped.

FabioM3, 2024-12-16:
I saw that the sterr.log have an error.

GregValiant, 2024-12-16:
If "Automatic Slicing" is turned on then yes, there can be a crash if it resets at a "bad" point in it's execution.
This might be a duplicate of #19904 and there is some discussion there.
For now I will mark this as a duplicate.

I think if you allow "Automatic Slicing" to finish before making a quick change to a Cura setting (which would force an immediate re-slice while something else is running) then you can mitigate the behavior.  The other option for now is to turn off "Automatic Slicing".

I'll leave this open as it might not be quite the same as the other one.

FabioM3, 2024-12-16:
I have tested more and for sure is the 'Auto slicing' + Preview tab that is the culprit.
If I change the script setting and slice manually nothing happens, but as soon as I turn on Auto slice in the first change in the script it crashes.
Even if I wait for a 10 seconds after the slicing is done it still crashes.

GregValiant, 2024-12-16:
This one and 19904 are linked together now.  I'll go ahead and close this as the other one was first and has more discussion.
You can keep up with any progress on the older bug report.

closed 2024-12-16 by GregValiant