Ultimaker/Cura #20043: Unable to add any non-networked printers.

Innovate3D, 2024-12-17:
### Cura Version

5.9

### Operating System

Windows 11

### Printer

Elegoo Neptune 3 Pro

### Reproduction steps

1. I tried to add another printer in cura since I recently got the Elegoo Neptune 4 plus.

### Actual results

It pulled up the screen like normal, but it didn't let me add any printers.

### Expected results

It should let me add another printer profile.

### Add your .zip and screenshots here ⬇️

[image]

Innovate3D, 2024-12-17:
> Do you perhaps have the "Startup Optimizer" plugin installed?
>
> If so, it may have prevented the Neptune 4 plus definition from loading,
> because you were not using it before. Go to Extensions -> Startup
> Optimiser -> Restore all configuration files, and restart Cura. Then try
> adding your printer again. Optionally, Go to Extensions -> Startup
> Optimiser -> Disable loading unused configuration files afterwards to
> reapply the startup optimisation.

Thank you this worked

GregValiant, 2024-12-19:
Thanks @fieldOfView .  I'll go ahead and close this.

closed 2024-12-19 by GregValiant