Ultimaker/Cura #20044: slicing failed

Galaxy-Prints, 2024-12-17:
### Cura Version

5.9.0

### Operating System

windows 10

### Printer

anycubic i3 mega X

### Name abnormal settings

no

### Describe model location

no

### Describe your model

no i downloaded it from thingiverse

### Add your .zip here ⬇️

[AI3MX_drawerbox_hex_v5.zip](https://github.com/user-attachments/files/18171672/AI3MX_drawerbox_hex_v5.zip)

GregValiant, 2024-12-18:
Thanks for the report and the file.
There was a bug in 5.6 that caused slicing to fail when there were a lot of walls on the "floor" of the print.  A couple of the models were like yours with honeycombs on the floor.  There were also problems with rough stone floors and rough hewn wood plank floors.
You seem to be experiencing a variant of that bug.  We will need the Cura.log file.

Re-create the problem and then use the "Help | Export Package for Technical Support" command.  Zip the file and post the zip folder here.  Along with the model and your settings that 3mf will also contain the Cura.log file.
The Cura team will take a look.

I didn't have any trouble slicing it (Win 10 Pro for an Ender 3 Pro) so I can't really duplicate this myself.

Galaxy-Prints, 2024-12-18:
Hello unfortunately i can no longer recreate the problem however I've
managed to fix it by re-installing the application and that seamed to work
but thanks for your help and support

GregValiant, 2024-12-19:
This might have been almost anything then.
I'm going to remove the bug label and close this as there isn't anything else to be done.

closed 2024-12-19 by GregValiant