Ultimaker/Cura #20047: Adding non-networked profile for Qidi X-Max3 cra

bowcoastie, 2024-12-18:
### Cura Version

5.9.0

### Operating System

Windows 10

### Printer

Qidi X-Max3

### Reproduction steps

1. Downloaded latest Cura.
2. Clicked Add Printer
3. Selected non-Ultimaker printer
4. Selected non-networked printer
5. Selected Qidi from drop-down list
6. Selected X-Max3 from list
7. Clicked add

### Actual results

Nothing happens. The program doesn't do anything. When I tried the other Qidi printer profiles, they continued on to install the corresponding printer. The X-max3 profile will not load.

### Expected results

I expected that the Qidi X-Max3 printer profile would be installed.

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2024-12-19:
Thanks for the report.
This has come up previously.  "QIDI" printers are not supported by UltiMaker Cura.  QIDI has their own fork of Cura but there are likely unknown file compatibilities between the QIDI version and the UltiMaker version.  Simply copying and pasting the definition files from QIDI Print (or any 3rd party app) into UM Cura really can't be expected to work.

I'll go ahead and remove the bug label and close this as it isn't a UM Cura problem.  QIDI print has diverged from UM Cura and variable names may have been changed, added, or deleted.  I hope you understand.

closed 2024-12-19 by GregValiant

bowcoastie, 2024-12-19:
Thank you Greg. I have contacted Qidi to see if they can help. If they give a workable response, I'll report back.

GregValiant, 2024-12-19:
It looks like QIDI forked UM Cura 4.9.1.  That version had around 600 settings and UM Cura 5.9.0 has 670 settings.  So a lot of settings have been added and some were deleted.  In addition, QIDI likely added there own machine-specific settings.

You can install a "Custom FFF" printer in UM Cura and once it's installed you can change the name and the Machine Settings to match your printer.  With QidiPrint open to Machine Settings your can go back and forth between the apps so that build plate size, firmware flavor, retraction distance and other machine specific settings all match.

My guess is that QIDI will respond with "just use QIDI Print".