Ultimaker/Cura #20048: Crashing when setting print parameters

Paul-Harder, 2024-12-18:
### Cura Version

5.9.0

### Operating System

Windows 11 Pro

### Printer

Creality CR 10S Pro

### Reproduction steps

When I try to change a setting in the print parameters the program crashes. 

### Actual results

Program shuts down and I restart it

### Expected results

Navn på fejlbehæftet program: CuraEngine.exe, version: 1.0.0.0, tidsstempel: 0x67374020
Navn på fejlbehæftet modul: ucrtbase.dll, version: 10.0.26100.1882, tidsstempel: 0x52db7a2d
Undtagelseskode: 0xc0000409
Fejlforskydning: 0x00000000000a502e
Id for fejlbehæftet proces: 0x6F1C
Starttidspunkt for fejlbehæftet program: 0x1DB515E5B15EBD4
Sti til fejlbehæftet program: C:\Program Files\UltiMaker Cura 5.9.0\CuraEngine.exe
Sti til fejlbehæftet modul: C:\WINDOWS\System32\ucrtbase.dll
Rapport-id: 2b3fa300-1668-413d-a3f6-e30d76ffc664
Fejlbehæftet pakkes fulde navn: 
Id for fejlbehæftet pakkerelativt program: 

### Add your .zip and screenshots here ⬇️

Log file - sorry its in Danish:
Navn på fejlbehæftet program: CuraEngine.exe, version: 1.0.0.0, tidsstempel: 0x67374020
Navn på fejlbehæftet modul: ucrtbase.dll, version: 10.0.26100.1882, tidsstempel: 0x52db7a2d
Undtagelseskode: 0xc0000409
Fejlforskydning: 0x00000000000a502e
Id for fejlbehæftet proces: 0x6F1C
Starttidspunkt for fejlbehæftet program: 0x1DB515E5B15EBD4
Sti til fejlbehæftet program: C:\Program Files\UltiMaker Cura 5.9.0\CuraEngine.exe
Sti til fejlbehæftet modul: C:\WINDOWS\System32\ucrtbase.dll
Rapport-id: 2b3fa300-1668-413d-a3f6-e30d76ffc664
Fejlbehæftet pakkes fulde navn: 
Id for fejlbehæftet pakkerelativt program:

GregValiant, 2024-12-18:
In Cura - use the "Help | Export Package for Technical Support" command.  Zip that file and post it here.
Hopefully the log file will contain the information from a previous crash.

GregValiant, 2024-12-19:
Thanks for that.
I'm not great at reading the log files but nothing obvious is jumping out.  The Cura team will take a look.

FYI - There was a previous bug report regarding a Win11 system just a day or so ago.  After installing a Windows upgrade package everything started working again.  Cura wasn't the only app affected on that person's system.

Paul-Harder, 2024-12-19:
Thank you Greg!
I did just update Windows 11 a few days ago but I have no other issues. I appreciate your efforts.
Kind regards, 
Paul