Ultimaker/Cura #20050: Build plate not loading when program opened

MadeInAK, 2024-12-19:
### Cura Version

5.9.0

### Operating System

Windows 11

### Printer

Ender pro 3

### Reproduction steps

Downloaded the program. First time opening it build plate was there and was able to slice as normal. Closed program. Next time I opened it build plate would not load. Uninstalled and installed again. Build plate would load first opening but after closing program build plate will not load
[image]

### Actual results

Build plate wont load

### Expected results

Build plate should load

### Add your .zip and screenshots here ⬇️

[image]

MadeInAK, 2024-12-19:
Windows had an update that needed to be installed - seems to be fix after updating windows

GregValiant, 2024-12-19:
Thanks for the report.
I'll leave the bug label on this (in case it comes up again) but I'll go ahead and close this as it seems to have been a Windows issue.

closed 2024-12-19 by GregValiant