Ultimaker/Cura #20058: Filament change post-processing not retracting

JucaNavazReque, 2024-12-21:
### Cura Version

5.8

### Operating System

Windows 11

### Printer

Creality Ender 3 Pro

### Reproduction steps

1. Load a file.
2. Add a post-processing script for filament change at certain layer (higher than 1).
3. Slice and print.

### Actual results

When the printer reaches the selected layer(s), it moves to (0, 0, Z) without moving upwards, moving through any plastic that is in between. Also, the plastic is not retracted, which most of the times leads to leaving a blob behind. When resuming the print, once again, the nozzle crosses through any plastic, leaving a subtle scar behind, and dragging the plastic blob.

### Expected results

The nozzle should retract the plastic and move a little upwards before moving to (0, 0, Z) to avoid this behavior.

### Add your .zip and screenshots here ⬇️

Not relevant.

GregValiant, 2024-12-21:
Thanks for the report.
"Filament Change" uses the M600 command which is firmware dependent.  So long as Cura added the line it is assumed that the printer will understand it.  (My own Ender 3 Pro does not have M600 enabled in the firmware.  I have to use PauseAtHeight instead.)

Does the M600 work on your printer?  Does the M600 command in the gcode have the "E" and "Z" parameters?

GregValiant, 2024-12-24:
_"I would recommend setting a default value greater than 0"_
Nothing moves fast around here but you can make that change yourself locally.
Go to your Cura installation folder (mine is "C:\Program Files\UltiMaker Cura 5.9.0\share\cura\plugins\PostProcessingPlugin\scripts" and open "FilamentChange.py" in a text editor.
Change line 93 from ""default_value": 0," to ""default_value": 1," and save the file.
That change will stay in place until you upgrade Cura.  You could also copy that file to your Configuration Folder and the "scripts" sub-folder.  That is where customized post-processors reside (but Cura may only list the one in the installation folder.  You'll have to check that.)

I think having to input multiple comma delimited settings in multiple text boxes is just confusing to a user.  If you have more than one filament change, and you need different settings for it, then enable another instance of "Filament Change".