Ultimaker/Cura #20070: [5.10.0]Cura Engine Build Fails on Windows Due to Missing pthread.h during Conan Installation

FreeFallingSnow, 2024-12-27:
### Cura Version

5.10.0-alpha.0

### Operating System

Windows 11

### Printer

during develop

### Reproduction steps

environment:
[image]
[image]

1. Follow the guide "Running Cura from Source" from the [Ultimaker/Cura Wiki](https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source) to run Cura from the source code.

2. In the "3. Initialize the Development Environment" step, run the following command:

   ```bash
   conan install . --build=missing -g VirtualPythonEnv -c user.generator.virtual_python_env:dev_tools=True -g PyCharmRunEnv
   ```

### Actual results

During the `conan install` process, the following error occurs in the CuraEngine build:

- A C++ compilation issue arises, particularly involving `std::construct_at`, with references to `cura::SupportInfillPart`, `coord_t`, and `bool` types.
- The build process halts with the error: `ninja: build stopped: subcommand failed`.
- In the `CMakeError.log` file found in the `CuraEngine/build/Release/CMakeFiles` directory, the following error is logged:

```text
Determining if the include file pthread.h exists failed with the following output:
Change Dir: C:/Users/username/.conan2/p/b/curae2689010b89b40/b/build/Release/CMakeFiles/CMakeTmp

Run Build Command(s):D:/GraduationDesignProgram/Path_planning/.venv/Scripts/ninja.exe cmTC_140d4 && [1/2] Building C object CMakeFiles\cmTC_140d4.dir\CheckIncludeFile.c.obj
FAILED: CMakeFiles/cmTC_140d4.dir/CheckIncludeFile.c.obj
D:\Work\VisualStudio\VC\Tools\MSVC\14.42.34433\bin\Hostx64\x64\cl.exe  /nologo   /DWIN32 /D_WINDOWS  /Zi /Ob0 /Od /RTC1 /showIncludes /FoCMakeFiles\cmTC_140d4.dir\CheckIncludeFile.c.obj /FdCMakeFiles\cmTC_140d4.dir\ /FS -c C:\Users\username\.conan2\p\b\curae2689010b89b40\b\build\Release\CMakeFiles\CMakeTmp\CheckIncludeFile.c
C:\Users\username\.conan2\p\b\curae2689010b89b40\b\build\Release\CMakeFiles\CMakeTmp\CheckIncludeFile.c(1): fatal error C1083: Cannot open include file: 'pthread.h': No such file or directory
ninja: build stopped: subcommand failed.
```

### Expected results

The development environment should be built successfully, allowing you to run `python cura_app.py` within the Cura Python virtual environment.

### Add your .zip and screenshots here ⬇️

[log.zip](https://github.com/user-attachments/files/18258153/log.zip)

zebin-xlogic, 2024-12-30:
The default encoding in the Chinese version of Windows is GB2312. To change it to UTF-8 encoding.

FreeFallingSnow, 2024-12-30:
> The default encoding in the Chinese version of Windows is GB2312. To change it to UTF-8 encoding.
Thank you. The problem has been solved.

closed 2024-12-30 by FreeFallingSnow

hzzlaystepup, 2025-03-23:
> > The default encoding in the Chinese version of Windows is GB2312. To change it to UTF-8 encoding.
> 
> Thank you. The problem has been solved.

hello，i met this problem ether,teach me how to solve it please,thank you!

FreeFallingSnow, 2025-03-23:
> > > The default encoding in the Chinese version of Windows is GB2312. To change it to UTF-8 encoding.
> > 
> > 
> > Thank you. The problem has been solved.
> 
> hello，i met this problem ether,teach me how to solve it please,thank you!

https://blog.csdn.net/robinhunan/article/details/106047345