Ultimaker/Cura #20073: Slicing stops working and starts back radomly for some tries and stops again

halevalente, 2024-12-27:
### Cura Version

5.9.0

### Operating System

Windows 11

### Printer

Kingroon KP5L (Using modified KP3S profile)

### Name abnormal settings

No abnormal settings, Doesn't matter the settings, Cura just stops slicing radomly.

### Describe model location

Rotating or moving have no effect.

### Describe your model

Model is watetight, never been able to slice it. The beahavior persists even with models I was able to slice before.

### Add your .zip here ⬇️

[Chest_Mimic_Toothy_01.zip](https://github.com/user-attachments/files/18264238/Chest_Mimic_Toothy_01.zip)
Hello, folks. Since Cura updated to 5.9.0, I have been unable to consistently slice models, experiencing the error shown in the attached image. Slicing randomly starts working again for one or two tries, then the behavior returns. I have tried reverting the print settings back to the factory defaults, but to no avail. The symptom now persists even if I try slicing the models on earlier Cura versions installed on my PC. Could you guys help me shine some light on this trouble?

halevalente, 2024-12-27:
The cura.log file and PC specs can be found inside the zip file attached.
[cura.zip](https://github.com/user-attachments/files/18264315/cura.zip)

I think also would be interesting mentioning that persisting on clicking the slice button to try again have caused some issues like BSODs, displaying many different errors, sometimes even related to Riot Games Vanguard application, and freezing of the SO, having to restart the PC manually.

GregValiant, 2024-12-28:
Thanks for the report.
This is a report on the model from formware.co/OnLineStlRepair
--> 0 Naked edges  (?)
--> 0 Planar holes  (?)
--> 0 Non-Planar holes  (?)
--> 1 Non-Manifold edges  (?)
--> 2 Inverted normals  (?)
--> 0 Duplicate faces  (?)
--> 0 Degenerate faces  (?)
--> 0 Disjoint shells  (?)
-> Repairing: 100.00%
-> Converting back
-> File repaired
Time needed for repair (ms): 7118
--> Vertex count changed from 62636 to 62366 (-270)
--> Triangle count changed from 125354 to 124818 (-536)
Sometimes errors don't matter and slicing will continue, but sometimes they do matter.
In this case I didn't have any trouble slicing the project on my Win10 system.  There have been some problems reported on Windows 11 systems.
[image]

From your log file is:
"Backend exited abnormally with return code 3221225477!"  my understanding is that is a generic error code.  Your system also tried to inform UltiMaker of the problem.  Hopefully they got the message.
"2024-12-27 20:21:27,621 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2024-12-27 20:21:27.621] [info] Sentry url: https://734f9ec9024f73e53701d59c3ffddfe3@o323038.ingest.sentry.io/4506257745510401"

Someone from the Cura Team will take a look.

halevalente, 2024-12-28:
Thank you for the quick answer. Hope for hearing from you guys soon. Meanwhile, I'll be checking my STL models and I'll report back if something new appears or if there are any improvements.

halevalente, 2024-12-28:
Performed some test with this same model, but fixed esing formware.co/OnLineStlRepair, with similar results as before. At first it seemed to be working, but then slivcing started to fail againg and I was never able to rotate, scale or do anything to the model, it would just crash and even BSOD my SO once. Hope this additional info to be useful. Thanks again.

HellAholic, 2024-12-30:
If you have a BSOD, then your OS/System is the source of the issue rather than the software you're running. Cura does not even have admin level access let alone kernel level to cause a BSOD. To check, just download the xyz calibration Cube and scale it up until it's 10mm short of your max build volume, then press slice. If you get a BSOD from this then check your CPU temps, in case you have 13th / 14th gen intel, check if your bios has been updated to the latest version. You can run a different CPU benchmark to verify the health state of your system.

halevalente, 2024-12-30:
[image]

Thank you for your quick response. I followed your suggestion and performed the slicing test with the XYZ Calibration Cube, scaling it to the maximum indicated build area, and it was successful. Additionally, there were no unusual temperature changes for either the CPU or GPU, as you can see in the attached image.

I also conducted further tests, and slicing always works in cases where support generation is not required. However, when supports are needed, I am unable to slice models larger than 60mm in height.

I’ll run some more extensive testing and double-check my GPU, CPU, and motherboard manufacturers' apps for any driver and/or firmware updates available. I’ll get in touch if anything unusual comes up or if I uncover additional details about the issue.

Thank you again for your assistance, and I appreciate your support.

halevalente, 2025-01-04:
Hello, guys,

Thank you for your assistance and support. Since my last message, I discovered that there were driver updates available for my motherboard and its components. I installed all the updates, and since then, I haven’t experienced any critical errors during slicing for the past few days.

Occasionally, I still need to cancel and restart the slicing process because the progress bar doesn't move for a while. Additionally, some support generation results come out a bit strange, but I’ve been able to resolve this by repositioning the model. Other than that, everything seems to be working fine.

Thank you once again for your support!

Best regards.

closed 2025-01-25 by halevalente