Ultimaker/Cura #20074: Concentric support first layer generated concentric to innermost wall of support brim instead of to support area

Slashee-the-Cow, 2024-12-28:
### Cura Version

5.9

### Operating System

Windows 11

### Printer

Ender-3 V3 SE, affects any printer though

### Reproduction steps

1. Load a model which will require support.
2. Enable *Generate Support* 
3. Set *Support Pattern* to *Concentric*
4. Enable *Enable Support Brim*
5. Set *Support Brim Width* to something not so huge it'd fill the entire area anyway.

### Actual results

The lines for the first layer of the brim are generated the *Support Line Distance* inwards from **the innermost line of the brim**:
[image]
But because the support brim goes *inwards* that means the lines will be out of alignment with above layers:
[image]
That's with four layers shown - you can see that there's one group of lines that are just one layer, and another group which are three layers. The ones which are three layers have nothing on the bottom layer, so each layer will just print in midair and fall down immediately, resulting in the support failing.

### Expected results

Concentric support lines on the first layer should be generated concentric to the outer support wall line - the one which will exist on all layers. If this results in a support line that would be in the same place as part of the brim, don't bother with that line since it will be supported on the layers above anyway.

### Add your .zip and screenshots here ⬇️

[support_brim_concentric.zip](https://github.com/user-attachments/files/18266345/support_brim_concentric.zip)
You don't really need to use my example though, I've found this problem to occur reliably on all models I tested.
Shout out to "palker" on the Cura forum who posted asking for help and I figured out the problem :)
~ Your Friendly Forum Assistant Moderator Slashee

GregValiant, 2024-12-28:
Thanks for the report.
I can duplicate this.
For a workaround, it seems that there are some "Magic Numbers" that can fool the bug.  Setting the "Brim Width" to an even whole number (2, 4, 6, 8) seems to work.

The Cura team will take a look.

This with the brim width at 2.0
[image]

Slashee-the-Cow, 2024-12-28:
Are you sure your "magic numbers" aren't just multiples of the support line
distance?

Slashee-the-Cow, 2024-12-29:
Also based on the example file of the problem palker posted on the forum using a multiple of the support line distance isn't a great solution for anything which isn't a basic shape because it looks like the concentric lines on the bottom layer were drawn by a five year old:
[image]
(That's showing two layers so you can see the discrepancy)