Ultimaker/Cura #20075: CURA 5.8 will not see my wifi Ultimaker3

daparcher, 2024-12-28:
### Cura Version

5.8

### Operating System

MacOS Sequoia

### Printer

Ultimaker 3

### Reproduction steps

Reconfigured my Ultimker e Printer to a wife using a new printer name but CURA does not list that new nema.  I see the original name, but it will not allow remova or renaming and when I slice I only see "Save to Disc" and not "Print over Network.

### Actual results

I can connect to the printer's camera through the I.P. Address in Digital Factory however.

### Expected results

I want to click the "Print over Network"  button to initiate printing.

### Add your .zip and screenshots here ⬇️

[image]

HumbleDeer, 2025-01-07:
You can add a second, new printer profile as a dummy in order to remove the Ultimaker profile.

Include your log file or the zip file generated, so we and/or developers can see what your Cura is reporting in the background.

daparcher, 2025-01-07:
Hi Anna .. thanks for the reply.  I refreshed the printer name via I.P.
address on the printer and am successfully connected to my wi-fi.  I still
have the issue, however, that after slicing I only get the Print To Disk
button.,  My printer's name is displayed on the Cora home interface but I
go to Manage Printers, hit Refresh or Add buttons --  it asks for an I.P.
address.  When I enter my printer's IP, nothing happens.
I
I've endered the printer's I.P. address intp my.browzer, the Ultimate pages
comes up but throws an error when I click on "Digital Factory" (see
attached).  In fact all the links on that page throw the same error.

I've rebooted the Macbook and the Printer ... (not the router though).

Yikes ...

[image: Screenshot 2025-01-07 at 11.52.29 AM.png]
[image: Screenshot 2025-01-07 at 11.51.46 AM.png]

daparcher, 2025-01-07:
UPDATE ..... I changed the wi-fi instance and am now able to access the
"Digital Factory" just fine .. I can see my printers, camera, logs, etc.
but CURA 5.9 will still only allow "Print To Disk" after slicing.

What can we do next?

Best,

HumbleDeer, 2025-01-17:
> Good morning. I still cannot print to my wi-fi network from Cura 5.9. I
> can connect to my U3 Digital Factory, camera etc, just fine by using the
> printer's IP address.
>
> In Cura however, after slicing any simple object, I still only get the
> "Print To Disk" option.
>
> I've included the export package for tech support.
>
> Please respond
>
>
> Best,
>
> Doug Parcher

Here's another possible troubleshooting step:

Set up a hotspot on your phone, tablet, or wirelessly connected device of
another type. Connect both your computer that runs Cura, as well as your
printer to it. Start with the printer removed from Cura, and restart the
computer before trying again to ensure any unexpected issues with ARP
tables or cached tables are not going to interfere.

Does it work now? If yes, the problem is definitely not with Cura.

HumbleDeer, 2025-01-17:
PS: I suggest you refrain from using "please respond" or liketype remarks,
as they can fall in a pretty bad taste. It's generally not very
appreciated, and that's not unique to open source slicers or printing
stuff. The type of remark may or could come across to someone as demanding,
hurrying, pressuring, or otherwise signalling discontent with the priority
or lack thereof you've been given for your ticket, or discontent with the
method or way time is handled across different people or subjects. Managing
time is important to prevent programmer burnout.

Good luck; HD

daparcher, 2025-01-18:
My bad, meant no harm.  Rough week.

HumbleDeer, 2025-01-18:
> My bad, meant no harm.  Rough week.
> […](#)

All good 🙂

daparcher, 2025-02-01:
Hi Anna ... FYI ... after migrating my Macbook Air M3 to Sequoia 15.1, then
15.2 and now 15.3, the anomaly is no longer (can print to the wi-fi
network).

Thanks ...