Ultimaker/Cura #20076: Multiple versions of cura on MacOS create duplication issues in System Settings/Privacy & Security/Local Network

thatguy-jaysenodell, 2024-12-28:
### Cura Version

Confirmed 5.6+ likely all

### Operating System

MacOS 15.2

### Printer

Crealty Ender 3 S1 Pro fronted by Octoprint

### Reproduction steps

1. install versions of Cura 5 in /Applications with each itteration being "UltiMaker Cura 5.x" where x is the minor version.
2. Update to 15.2
3. Attempt to print to Octoprint fronted printer... octoprint no longer found. 
4. In System/Privacy & Security/Local Network there are 2 entires for "UltiMaker Cura 5"
5. Delete all cura versions.
6. reboot
7. reinstall cura 5.9
8. Everything works
9. reinstall 5.7 2 entires appear and octoprint is gone. 
10. Multiple itterations of testing and now I'm unable to remove the "UltiMaker Cura" from entries. 

### Actual results

Multiple entries in System Settings/Privacy & Security/Local Network

### Expected results

Based on https://developer.apple.com/forums/thread/766270 this is likely due to using naming for multiple instances that can not be "unique" enough in the Privacy & Security DB space. I am not sure this is an addressable "bug" but should certianly be reviewed and updated in documentation as the "5.6/5.7/5.8/5.9" naming context is common to allow multiple instance of cura to coexist. 

### Add your .zip and screenshots here ⬇️

I have documented this on the ultimaker community forum as well: https://community.ultimaker.com/topic/46501-cura-59-67-under-macos-152-not-connecting-to-octoprint/

thatguy-jaysenodell, 2024-12-28:
As a note, report 20075 is oddly similar to what I'm seeing but with a full ultimaker stack. It might be worth verifying if that user has multiple versions of cura installed.

dsu-igeek, 2025-01-30:
This may be connected to https://github.com/fieldOfView/Cura-OctoPrintPlugin/issues/331.  Turning local network access off and on for Cura seems to fix the issue for a while.