Ultimaker/Cura #20078: Infill inside corners and non-sharp corners

sargonphin, 2024-12-29:
### Cura Version

5.9.0

### Operating System

Windows 11

### Printer

Heavily modified Tronxy X5SA Pro

### Reproduction steps

I am trying to print a box with walls that are close to a multiple of wall width but not quite, for instance a box with 2mm walls and wall lines thickness of 0.45

### Actual results

Wall line thickness is not respected (goes over 0.6mm for a 0.4mm nozzle), wall corners are not sharp and instead filled with infill and not printed square

[image]

Reducing the wall line width to an exact multiple of the box wall thickness results in one infill line printed in between the lines with still incorrect shape

[image]

These images are showing layer type, red is outer wall, green is inner wall and orange is infill

### Expected results

Walls are filled with infill or, preferably, with an extra wall line and makes a proper right angle instead of a curve as shown below

[image]

### Add your .zip and screenshots here ⬇️

Project file: [tronxyCoreXY_motorCoolers-GITHUB.zip](https://github.com/user-attachments/files/18268174/tronxyCoreXY_motorCoolers-GITHUB.zip)

GregValiant, 2024-12-29:
Thanks for the report.
A couple of weeks ago I performed a geometric analysis of the way that a couple of other slicers do their corners and the way that UM Cura does it's corners.  The difference in area between the multiple gaps left by the other slicers, and the single largish gap that Cura leaves, is near zero.  From a simple geometry standpoint there really isn't any difference.  The big difference is in the silly travel moves required by Cura when it has to make a lap of a model and leave little dots to fill in the corners.  That is a significant hit on printing time.

Although it isn't working the way you would like, it is working exactly as designed and so it isn't a "bug" and I'll remove the label.
This report would be good as a "Feature Request" to change the toolpath planning around corners.  You can change your write-up and I'll add the label.

Cura currently uses "Concentric" paths around inside corners while some others use parallel paths around inside corners.  Both of them leave gaps.
This is an inside corner.  There are 9 gaps with a square corner outside and a radius of 1/2 Line Width on the inside.
[image]

On an outside corner, the extrusions are concentric and so there are no little gaps, but there will be the characteristic gap in the middle.  Here I have the wall count high enough to fill it, and because they are all "inner walls", the extra travel doesn't occur.  If the wall count was lower then the gap would probably be filled with infill and that would cause the extra travel.
[image]

This is the same model in PrusaSlicer.  It looks sort of similar.  The variations between this and the Cura preview may well be in the screen representation rather than any real difference in the print.
[image]

sargonphin, 2024-12-29:
The "gaps" in the corners are mitigated by the fact that there is always some springiness in the filament, even easy ones like PLA, which will always over-extrude (ooze) the tiniest bit and fill these gaps. It becomes very evident when printing the first few layers, there is no physical gap because of the filament being squished out of the nozzle during direction changes fills in this empty space. This also results in a generally better quality print, as the filament has somewhere to go instead of only outside the wall (makes the protrusion better). Printing with a slicer that leaves these gaps open results in a much better quality corner

I would also like to add that, Cura needs to offer the possibility to generate classic walls instead of strictly Arachne. I am observing large gaps between wall lines in my model because the nozzle is not able to reliably extrude that much material. This results in separate wall lines which are very Fragile. It would be much, much better to use an extra thinner line than try to print large lines for quality purposes. Cura must offer a way to disable printing those excessively large lines

[image]
[image]

closed 2025-04-05 by sargonphin

GregValiant, 2025-04-05:
I've been printing a lot of carbon fiber filled PETG lately.  It's different as the fill makes the material more viscous and conseuently I print it 10° higher (245°) than regular PETG.  The increased viscosity affects the flow but I have not seen any problems with walls coming apart like you show there.

I read someplace that it is not possible to put a "upper line width" limit when using variable line width.  I don't remember the reasoning, but it made sense.
What you can do is increase the "minimum line width" to get different effects.  You can also adjust the "Flow Equalization Ratio" in the Speed settings.  A higher number will slow the print head so it has time to extrude the flow increase on those wide lines.

Looking at your images, all the wall extrusions appear to be very round instead of squished.  That makes be wonder if you are suffering from simple under-extrusion in the walls.  It could also be that the printing temperature is coming into play as those walls should really be bonded together.

sargonphin, 2025-04-05:
I will give it some further testing, I changed a lot of the settings since this print and it might be worth retrying to print this box just to see