Ultimaker/Cura #20079: Could not find plugin UFPWriter

DaveTPilot, 2024-12-29:
### Cura Version

5.9.0

### Operating System

Windows 11

### Printer

Creality CR6 SE

### Reproduction steps

1. When attempting to upload a sliced model via Moonraker, I get the following error message: **Could not find plugin UFPWriter**
2. I reinstalled Cura.
3. I deleted Cura and reinstalled Cura again.

### Actual results

Same results

### Expected results

Up until recently, Cura would upload via the UFP with Thumbnail via Moonraker. 

### Add your .zip and screenshots here ⬇️

[image]

fieldOfView, 2025-01-01:
Please share your cura.log. You can find it in the configuration folder (see Help -> Show configuration folder).

It would be my guess that you have somehow disabled the UFP Writer plugin in the Marketplace dialog (via the gear icon in the top right of that dialog).

DaveTPilot, 2025-01-16:
Yes. I actually solved this a couple of weeks ago. You are correct. The UFP
Writer was disabled. It was difficult to find it by searching for plugins.

fieldOfView, 2025-01-16:
Thanks for letting us know you fixed your problem.

closed 2025-01-16 by fieldOfView