Ultimaker/Cura #20082: Cura Slicer 3D View Not Functioning on Fedora Linux

FaFiii, 2024-12-29:
### Cura Version

5.4.0

### Operating System

KDE Plasma Fedora 41 Linux 

### Printer

Pursa i3

### Reproduction steps

1.     Install Cura Slicer on a Fedora Linux system.
2.     Launch the application.
3.     Attempt to load a 3D model into the workspace.

### Actual results

The 3D viewport remains blank, preventing interaction with the loaded model.
[image]

### Expected results

The 3D model should display in the central viewport, allowing for standard manipulation and slicing operations.

### Add your .zip and screenshots here ⬇️

[image]
[Archive.zip](https://github.com/user-attachments/files/18270193/Archive.zip)

GregValiant, 2024-12-29:
Thanks for the report.
There are a lot of errors in the log file regarding "OpenGL" that Cura uses to show the model.
```
2024-12-29 16:54:20,806 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:   File "/usr/lib/python3.13/site-packages/UM/View/GL/OpenGL.py", line 51, in __init__
2024-12-29 16:54:20,807 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:     raise RuntimeError("Try to create singleton '%s' more than once" % self.__class__.__name__)
2024-12-29 16:54:20,808 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: RuntimeError: Try to create singleton 'OpenGL' more than once
2024-12-29 16:54:20,815 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [66]: An uncaught error has occurred!
2024-12-29 16:54:20,816 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: Traceback (most recent call last):
2024-12-29 16:54:20,816 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:   File "/usr/lib/python3.13/site-packages/UM/Qt/Bindings/MainWindow.py", line 289, in _render
2024-12-29 16:54:20,817 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:     OpenGL()
2024-12-29 16:54:20,817 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:     ~~~~~~^^
```
Cura 5.4 is 7 versions out of date.  You may be having a video system problem with your "Open GL drivers".
Cura 5.9 stable is out.  Try upgrading Cura (the files are [HERE ](https://github.com/Ultimaker/Cura/releases/tag/5.9.0)under Assets) and see if it fixes the problem.

The Cura team will take a look.

FaFiii, 2025-01-12:
Thank you for this information. I installed the latest version of Cura (5.9) using the provided AppImage, and it's working much better. Nevertheless, my issues are resolved, so I am closing this bug.

Kind regards,
FaFiii

closed 2025-01-12 by FaFiii