Ultimaker/Cura #20086: Temperature settings not working with MethodX

MPuharinen, 2024-12-30:
### Cura Version

5.9.0

### Operating System

Win10

### Printer

Ultimaker MethodX

### Reproduction steps

Slicing done with LABS extruder and support extruder. Material selected Ultimaker ABS, in Material settings printing temperature was set 220°C. Print file saved to removable disk as .makerbot file. When starting to print, I noticed that printing temperature was 245°C instead of the custom set 220°C. 245°C is default temperature for Ultimaker ABS.

### Actual results

Printing temperature is not correct, somehow it takes the default even in my case better result is with lower temperature.

### Expected results

Expected to get custom set temperatures.

### Add your .zip and screenshots here ⬇️

[UMMX_SW00001334_F_Plug_2.5mm.zip](https://github.com/user-attachments/files/18274406/UMMX_SW00001334_F_Plug_2.5mm.zip)

GregValiant, 2024-12-30:
Thanks for the report.
Integration of the MethodX printers remains a work in process and it is being worked on.
Maybe @rburema is the right person to ask about this?

HellAholic, 2024-12-30:
Short answer, you need to duplicate the ultimaker ABS material and set a custom printing temperature for the material, the machine only cares about the temperature specified by the material when heating up the nozzles and for the default printing temperature.
I assume the Method team has a reason for setting up the configurations in this manner, but if you would want to change it as a user and accept the associated risks, you can do so by:

Preferences > Configure Cura > Materials > Select Ultimaker ABS
[image]

Click on the burger menu and press duplicate, select a Display name to identify the material with lower temp (optional)
[image]
Go to the print settings tab > Set default printing temp at 220
[image]
Activate the newly added material
[image]
Your nozzle should now heat up to 220.

MPuharinen, 2024-12-31:
Thanks,

Now, I got this. I asked this earlier from  ultimaker support, but they didn’t told me this.

BR,
Marko

image.png (view on web)<https://github.com/user-attachments/assets/91589f7d-b60e-4dd5-9754-ad28ceafb539>

image.png (view on web)<https://github.com/user-attachments/assets/5d78eda4-1d52-4a2a-b92a-7ebda41e34dc>
image.png (view on web)<https://github.com/user-attachments/assets/2cd24233-3510-4c7e-abb0-26b5b1ccc93d>
image.png (view on web)<https://github.com/user-attachments/assets/b2cb5c19-769a-41b0-8238-b730878ab14e>