Ultimaker/Cura #20087: Letting us put rafts under supports would be nice

fast2611, 2024-12-30:
### Is your feature request related to a problem?

Support adhesion with the current crisscrossing grid or internal brim can be quite poor, especially with small tree support bases.

### Describe the solution you'd like

It'd be a huge help if we could set the bottom pattern of the support  already, but what would really be a game changer would be the ability to give them a small raft with a bit of an extended edge instead of the current internal only brim

### Describe alternatives you've considered

increasing Support infill density multiplier - too flexible and nocks itself loose
solid  internal brim: warps alongside the edge and pops off

### Affected users and/or printers

I've run into this more on printers with PEI sheets

GregValiant, 2024-12-30:
An option for the brim to develop on the outside of the tree trunk would help.
The amount of brim area is dependent on the trunk diameter so smaller trunks have very limited contact with the build surface.

The new higher printing speeds on bed slingers isn't helping either.  It's easy for the print to break loose from the bed when the starts and stops are very sudden.

HellAholic, 2024-12-30:
If you enable the build plate adhesion set it to brim > set brim location to outside only > and give it a reasonable width/brim line count, it will generate Outside Brim for the tree support.
The drawback is that your model will also get a brim.
If an outside brim resolves the issue it will be simpler to add a "Brim inside/outside/both" setting for the support than generating a raft for the support.

fast2611, 2024-12-30:
> If you enable the build plate adhesion set it to brim > set brim location to outside only > and give it a reasonable width/brim line count, it will generate Outside Brim for the tree support. The [...]

yes. printing and having to remove a 40x40cm multi layer raft under a model that sticks to the build plate just fine as is with a perfect surface finish just so two 1cm thick towers dont rip off on the 6th layer is what I would call a downside. I just had the joy of doing that after finding there is no alternatives, which is why I created this request.

fast2611, 2024-12-30:
... this is the second time I fatfinger the close button, sorry!

HellAholic, 2024-12-30:
Rafts overall are just messy, my suggestion was to enable Brim as the build plate adhesion rather than a raft.
If Brim on the outside of your support can resolve the issue (5 lines of brim for example or any other number that gives stability) and the support is stable enough with it, then it's a much easier and faster feature to prioritize and implement.
Compare this (Build plate adhesion set to Brim) -> could solve the problem, if so it's much easier to implement and cleaner
[image]
with this (Build plate adhesion set to raft) -> solves the problem but takes extra time and is messy
[image]

fast2611, 2024-12-30:
I dont really see your point about it being "messy". Either way the ahesion works alot better with a stiffer multi layer sheet, and the implementation of having adhesion helpers outside of the supports perimeter while not having them on the model itself would need to be implemented in both cases

HellAholic, 2024-12-30:
Printing a raft will take at least 3x the time of printing a brim for the same area.
Brim has 4-5 settings to configure and raft has 58 settings.
Support is a throw-away part, so the less time / material you spend on it the better.

spiritparticle, 2024-12-31:
It is such a strange thing where people all over the internet will criticize something due to their personal "taste".
Who cares if it is "messy". What does that even mean? It is a feature for a functional aspect of not only increasing stability but also adhesion where a brim might fail and has happened many times to me before where even with a wide brim the part will just come off and I lose my supports. Out of frustration I made a brim so large it literally connected to evey part of the base of the model and ended up probably using as much time and filament if there was simply a raft there. Which brings me to my next point. Someone saying it will take 3x the time... So what? I would GLADLY give up that time to ensure the security of supports and justify the extra time for the extra security to not only ensure the print will not fail and waste my filament and then I have to print again and work out a way to making the supports better adhere and even resorting to the old trusty glue stick. Just having the investment on 3x time spent on a sturdy raft for the supports to save the possibility of the hours sometimes DAYS spent printing something only to start again because a tall support came lose before it got to the overhang.

It is a justified and frankly useful feature request that is getting lost because people here have their own personal taste on how something should look or how long they want their print to take. Those are PERSONAL preferences. If you don't want to "waste" time on a raft for a support. Don't use it. If you think it looks "messy" then don't use it.

It is such a strange thing to me that a legitimate feature request because something like a brim is unreliable. Gets people posting their two cents suggesting brims which are unreliable to a degree. Then also posting how they won't like how it "looks"

Why does that even matter? I would LOVE this feature to happen. I would like to +1 this idea. Maybe even add to it that little triangle shaped joints be added to the base of the trees to offer additional structural support.

When printing a model 30cm high... The higher the support gets the more unstable the brim at the base gets. This is a great idea.

HellAholic, 2024-12-31:
@spiritparticle It matters when the person who's talking about it is part of the Cura team developing the actual software you're making the feature request for. So yes, when I say MESSY, it is messy, it's not a personal opinion, it's a deduction based on familiarity with the software. If you only care about the print finishing then just add a raft to all your prints.
This discussion is getting nowhere, so I'm out. Good luck.

GregValiant, 2024-12-31:
One thing we all must remember is that the Cura Team does the heavy lifting in regards to coding Cura. The team has opted out which leaves it up to "community members" to add this feature.
I'll mark this as a "won't do" and close it.

closed 2024-12-31 by GregValiant