Ultimaker/Cura #20089: ImportError: DLL load failed while importing QtNetwork:no module named 。

zebin-xlogic, 2024-12-31:
### Cura Version

5.9.0

### Operating System

Windows 11

### Printer

no

### Reproduction steps

1. **Activate the virtual environment located in the `Cura\build\generators\cura_venv\Scripts\Activate` directory.**
2. **Run the `cura_app.py` file**

### Actual results

Traceback (most recent call last):
  File "B:\cura_space\Cura\cura_app.py", line 27, in <module>
    from PyQt6.QtNetwork import QSslConfiguration, QSslSocket
ImportError: DLL load failed while importing QtNetwork: The specified module could not be found.

### Expected results

be able to run cura

### Add your .zip and screenshots here ⬇️

[image]

lik15, 2025-01-03:
I encountered the same problem when running Cura from the source code. Have you solved the problem？

zebin-xlogic, 2025-01-06:
I created a new env to run 'cura_app.py'.  The next step you can see that [#20084](https://github.com/Ultimaker/Cura/issues/20084)

closed 2025-01-06 by zebin-xlogic

pskowronskiTDx, 2025-02-18:
@wawanbreton I am currently running into the same issue. As I am not using the PyCharm, the solution provided in #20084 is not helpful for me. I don't know how to link Uranium, as the steps described in wiki appear to be not working now (guess it is due to upgraded version of conan). Can you provide any hints on that?

wawanbreton, 2025-02-18:
> [@wawanbreton](https://github.com/wawanbreton) I am currently running into the same issue. As I am not using the PyCharm, the solution provided in [...]

The main Wiki pages for running Cura from sources have been updated, but they are a bit fresh indeed. Can you tell me more precisely what step would not work for you ? Are you trying to use Uranium as an editable package ?

pskowronskiTDx, 2025-02-18:
Yes, exactly, it's about putting Uranium into an editable mode. The command "conan editable add . uranium/..." fails, complaining about unrecognized arguments.