Ultimaker/Cura #20090: Support wall line count=0 without objects erroneously homes on every layer

fast2611, 2024-12-31:
### Cura Version

5.9.0

### Operating System

Win 10

### Printer

Custom FFF

### Reproduction steps

Tree support with circular infil and "Support wall line count" set to 0, if the model floats completely over the build plate.
I consider this a low priority bug due to how extremely specific it is, but wanted to make sure to report it regardless

### Actual results

The printer homes to 0 0 on every layer change until the solid parts of the model are reached

### Expected results

it shouldn't home to 0 0 for no reason. The other behaviors from this weird combination of settings are desirable, so please don't touch those

### Add your .zip and screenshots here ⬇️

[example.zip](https://github.com/user-attachments/files/18281963/example.zip)
[image]

GregValiant, 2024-12-31:
Thanks for the report.
Your layer height of 0.12 and your "Support Infill Layer Thickness" of 0.50 seems to be causing the problem.
It kind of looks like the head is parking so it doesn't ooze over the print while it indexes up through 3 empty layers.  I don't see a setting that might address that.  Changing the "Support Infill Layer Thickness" to 0.12 fixes the travel problem as does adding a Support Wall.

The Cura team will take a look.  

This is in 5.8.0 and the problem doesn't seem to happen even at 0.5 Support Infill Layer Thickness.
[image]

fast2611, 2024-12-31:
I did not consider the "parking" due to empty layers, that does make sense. 
considering a bare bones support with no model filling the layers in between is not exactly common I reported this as more of a curiousity in case there might be a deeper issue, thanks for the insight!